ev3dev/ev3dev #682: No internet connection.

PyotrK, 2016-06-20:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3-ev3dev-jessie-2015-12-30.img.xz

* My kernel version is: 3.16.7-ckt21-9-ev3dev-ev3

* My host computer is running: Android 5.1

* I am connecting using: Bluetooth

### About my issue:
I can't connect to the internet, without using Android. Using Android is a struggle and i want to use my Mac (OS X 10.9.5). I have tried connecting using both bluetooth and a usb cable.The network connection does not change to online, but to connected. I am new to programming anything outside the ev3 home edition.

dlech, 2016-06-20:
Have you followed the tutorials?

* http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-bluetooth/
* http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/

PyotrK, 2016-06-20:
yes

PyotrK, 2016-06-20:
it is:
192.168.44.201

PyotrK, 2016-06-20:
Also forgot to mention that i do not have the CDC Composite Gadget option in the Networks thingy (i do not know what those menus are called).

dlech, 2016-06-20:
"EV3+ev3dev" is the same as "CDC Composite Gadget"

Is "192.168.44.201" from connecting to Android? What is it when connecting to Mac? I would expect it to be "192.168.**2**.x"

PyotrK, 2016-06-20:
Using usb changes the ip to:
169.254.77.202

Using blutooth:
192.168.2.3

dlech, 2016-06-20:
So with USB, "Sharing" is not setup correctly because this is a fallback IP address. Bluetooth looks correct though. I would expect you to have an Internet connection then unless there is some type of network problem that is blocking it.

PyotrK, 2016-06-20:
This is how the settings look:

[image]

dlech, 2016-06-20:
Try sharing only one at a time. If you are using USB, only check "EV3+ev3dev"

PyotrK, 2016-06-20:
I tried using only one. Still only connected.

This might be important but i don't know
[image]

Translation:
Status: Connected
EV3 + ev3dev has a self-assigned IP address and will not be able to connect to the Internet.

dlech, 2016-06-20:
It tells us what we already knew. "self-assigned" == 169.254.x.x

Unfortunately, I am out of ideas. These tutorials have always worked for me.

You could always try rebooting your Mac if you haven't tried that already.

PyotrK, 2016-06-20:
Reebooting now, will come back with info if it works or not.

PyotrK, 2016-06-20:
Still only connected :(

Any way of using Android as something that just forwards the connection to the ev3?

dlech, 2016-06-20:
I think I have used this program before to do that sort of thing:

https://play.google.com/store/apps/details?id=at.bherbst.net&hl=en

PyotrK, 2016-06-21:
I don't know how to use it, but i tried to rewrite the image. When i booted it up everything seemed normal, but then i got an error when trying to connect to the network via bluetooth:
Timeout was reached.

Any idea if this has been stopping it from connecting to the internet?