ev3dev/ev3dev #683: Follow most recent recommendations for sysfs show (no snprintf)

bmegli, 2016-06-23:
All the code I have seen is ok as it is now, this is just a cosmetic change. 

Issue to remind ourselves to follow the latest recommendations (changed in mid-2015):

From https://www.kernel.org/doc/Documentation/filesystems/sysfs.txt:
```
show() must not use snprintf() when formatting the value to be
  returned to user space. If you can guarantee that an overflow
  will never happen you can use sprintf() otherwise you must use
  scnprintf().
```

At least to apply in:
https://github.com/ev3dev/lego-linux-drivers/blob/036f41ab854c3cdc21da3b6451a94fa57d2d606a/sensors/lego_sensor_class.c
https://github.com/ev3dev/lego-linux-drivers/blob/master/include/tacho_motor_class.h

The reason is probably the fact that snprintf returns the number of bytes that `would be written` instead of  `that was written` and this caused a lot of errors in the past.

rhempel, 2016-06-23:
Well, I'm in the driver code now so I will make an effort to eliminate this use of snprintf() - thanks for pointing it our

bmegli, 2016-06-24:
Ok! It's not pressing, it's just to remember to do it someday

rhempel, 2016-06-24:
Thanks @dlech for adding the label - @bmegli I have a branch already prepared for the work so I don't forget to do it :-)

dlech, 2017-10-02:
fixed in `ev3dev-1.5.1`

closed 2017-10-02 by dlech