ev3dev/ev3dev #685: Program in ev3dev with blocks?

introlinux, 2016-06-25:
Hi, is there any way to program in ev3dev using some beginners environments like Snap! Scratch! or similar?

Thanks in advance.

mob41, 2016-06-25:
I am currently developing a [application](https://github.com/mob41/jEv3dev) to just program with blocks. However, it is still in very early development stage, it does nothing. I have to develop the GUI first before IO.

I highly recommend you to learn some programming language. C++, Java, Python. It is easy to learn. And there is lots of libraries that helps you to access to the motors, sensors directly. See [here](https://ev3dev.org/docs/libraries)

mob41, 2016-06-25:
Wow, I haven't figured out that ev3dev has Open Roberta.

closed 2019-03-25 by dlech