ev3dev/ev3dev #686: Brickmon program termination and motor reset

kortschak, 2016-06-26:
What signal is sent by brickmon to programs that are running when the back button is pressed? and related, what techniques do people use to ensure motors are stationary when a program terminates due to a signal or error (I've just written a ResetAll function for ev3go, but I was wondering if others have already established approaches).

dlech, 2016-06-26:
> What signal is sent by brickmon to programs that are running when the back button is pressed?

Nothing if the back button is pressed momentarily. If the back button is held down for [one second](https://github.com/ev3dev/brickman/blob/ev3dev-jessie/src/GlobalManager.vala#L93), then it [kills everything on the virtual console](https://github.com/ev3dev/brickman/blob/ev3dev-jessie/src/controller/FileBrowserController.vala#L104) (`SIGTERM`). However, since we are running the process as a subprocess of `openvt` and `su`, these may forcibly terminate your process before it has a chance to handle `SIGTERM`.

> what techniques do people use to ensure motors are stationary when a program terminates due to a signal or error

When running a program from brickman, when the user program ends we [reset motors, leds and sound](https://github.com/ev3dev/brickman/blob/ev3dev-jessie/src/controller/FileBrowserController.vala#L124). This way, even if the user program crashes, everything will stopped.

There is also #388 for a script that does basically the same thing that brickman does for running your programs from a remote terminal.

Even if you program handles `SIGTERM`, `SIGQUIT`, etc. you are still going to have a problem if your program crashes (e.g. `SIGSEGV`), so it is best to have a separate process that can monitor your program in case of a crash.

dlech, 2017-10-28:
In ev3dev-stretch, Brickman now uses `brickrun`, which sends `SIGKILL` when the back button is pressed and stops all motors, sound, LEDs, etc.

Basically, user programs don't need to worry about being interrupted or crashing. If you want to exit gracefully for style points, go for it, but it is not required.

closed 2017-10-28 by dlech