ev3dev/ev3dev #687: "hold" stop_action has paradoxical reverse action in some cases

kortschak, 2016-06-26:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3-ev3dev-jessie-2015-12-30.img

* My kernel version is: 4.4.13-12-ev3dev-ev3

* My host computer is running: Linux

* I am connecting using: Wi-Fi

### About my issue:

Run the code at [znap.go](https://github.com/ev3go/ev3dev/blob/f7d668b8473ad00c27b9ed21cec3d05dead34efc/examples/znap/znap.go) with the `brake` `stop_action` specified at [znap.go#L208](https://github.com/ev3go/ev3dev/blob/f7d668b8473ad00c27b9ed21cec3d05dead34efc/examples/znap/znap.go#L208) and [znap.go#L223](https://github.com/ev3go/ev3dev/blob/f7d668b8473ad00c27b9ed21cec3d05dead34efc/examples/znap/znap.go#L223) replaced with `hold`.

This program is a reasonable approximation of the Znap! model EV3-G code published by LEGO. When an object is placed within ~25cm of the US sensor the jaws (`outB`) extend and then return. On return, when their `stop_action` is `hold`, the large motors (`outA` and `outD`) run at full speed in reverse - this does not happen with the `stop_action` set to `brake`.

A log of motor states at ~200ms intervals shows this:
```
$ ./znap; ./stopall 
2016/06/26 08:26:41 outB: none 0
2016/06/26 08:26:41 outA: none 0
2016/06/26 08:26:41 outD: none 0
2016/06/26 08:26:41 outB: none 0
2016/06/26 08:26:41 outA: running 584
2016/06/26 08:26:41 outD: running -593
2016/06/26 08:26:42 outB: none 0
2016/06/26 08:26:42 outA: running 523
2016/06/26 08:26:42 outD: running -522
2016/06/26 08:26:42 outB: none 0
2016/06/26 08:26:42 outA: running 533
2016/06/26 08:26:42 outD: running -525
2016/06/26 08:26:42 outB: none 0

// provoke threat response - distant

2016/06/26 08:26:42 outA: running 519
2016/06/26 08:26:42 outD: running -525
2016/06/26 08:26:43 outB: running|ramping -143
2016/06/26 08:26:43 outA: running 533
2016/06/26 08:26:43 outD: running -525
2016/06/26 08:26:43 outB: holding 78
2016/06/26 08:26:43 outA: running 544
2016/06/26 08:26:43 outD: running -538
2016/06/26 08:26:43 outB: running|ramping 552
2016/06/26 08:26:43 outA: running 514
2016/06/26 08:26:43 outD: running -528
2016/06/26 08:26:43 outB: none 0
2016/06/26 08:26:43 outA: holding 99
2016/06/26 08:26:43 outD: running|ramping -372
2016/06/26 08:26:44 outB: none 0
2016/06/26 08:26:44 outA: running -547
2016/06/26 08:26:44 outD: running 586
2016/06/26 08:26:44 outB: none 0
2016/06/26 08:26:44 outA: running -552
2016/06/26 08:26:44 outD: running 526
2016/06/26 08:26:44 outB: none 0
2016/06/26 08:26:44 outA: running -528
2016/06/26 08:26:44 outD: running|ramping 469
2016/06/26 08:26:45 outB: none 0
2016/06/26 08:26:45 outA: running 540
2016/06/26 08:26:45 outD: running 437
2016/06/26 08:26:45 outB: none 0
2016/06/26 08:26:45 outA: running 521
2016/06/26 08:26:45 outD: running 521

// provoke threat response - near

2016/06/26 08:26:45 outB: running|ramping 162
2016/06/26 08:26:45 outA: running 527
2016/06/26 08:26:45 outD: running 521
2016/06/26 08:26:45 outB: running|ramping -487
2016/06/26 08:26:45 outA: running 530
2016/06/26 08:26:45 outD: running 533
2016/06/26 08:26:46 outB: none 16
2016/06/26 08:26:46 outA: running 524
2016/06/26 08:26:46 outD: running 527
2016/06/26 08:26:46 outB: none 0
2016/06/26 08:26:46 outA: running 529
2016/06/26 08:26:46 outD: running 529
2016/06/26 08:26:46 outB: none 0
2016/06/26 08:26:46 outA: running 504
2016/06/26 08:26:46 outD: running 525
2016/06/26 08:26:46 outB: running|ramping 378
2016/06/26 08:26:46 outA: running 540
2016/06/26 08:26:46 outD: running 512
2016/06/26 08:26:47 outB: running 632
2016/06/26 08:26:47 outA: holding|overloaded -419
2016/06/26 08:26:47 outD: holding|overloaded 512
2016/06/26 08:26:47 outB: running 770
2016/06/26 08:26:47 outA: holding|overloaded -789
2016/06/26 08:26:47 outD: holding|overloaded 821
2016/06/26 08:26:47 outB: running 794
2016/06/26 08:26:47 outA: holding|overloaded -793
2016/06/26 08:26:47 outD: holding|overloaded 830
2016/06/26 08:26:48 outB: holding 0
2016/06/26 08:26:48 outA: holding|overloaded -828
2016/06/26 08:26:48 outD: holding|overloaded 838
2016/06/26 08:26:48 outB: holding 0
2016/06/26 08:26:48 outA: holding -98
2016/06/26 08:26:48 outD: holding 94
2016/06/26 08:26:48 outB: holding 0
2016/06/26 08:26:48 outA: holding -27
2016/06/26 08:26:48 outD: holding -30
2016/06/26 08:26:48 outB: holding 0
2016/06/26 08:26:48 outA: holding 0
2016/06/26 08:26:48 outD: holding 41
2016/06/26 08:26:49 outB: holding 0
2016/06/26 08:26:49 outA: holding 0
2016/06/26 08:26:49 outD: holding -38
2016/06/26 08:26:49 outB: holding 0
2016/06/26 08:26:49 outA: running 583
^C
```

rhempel, 2016-06-26:
Please add a small (200 msec or so) `ramp_down_sp` to your motor motion block. That gives the driver time to ramp the speed down before it hits the `position_sp`.

Think about it - you are the motor and I tell you to walk 10 meters, but I don't tell you until you are at the 10 meter position to stop. You will probably have to go back a half step :-)

kortschak, 2016-06-26:
There is a ramp up and ramp down already present - set at the lines just above the linked lines.

Even if there were not, I don't think it would explain the degree of reverse. If you look at the log, you can see the motor being driven at full reverse for at least a full second, having been driven forward previously only at half speed.

kortschak, 2016-06-26:
In addition, the effect is not seen for every situation where a stop happens - this can also be seen in the log above.

rhempel, 2016-06-26:
@kortschak - does the model actually move backwards for a second? There is a defect in the current motor driver #661 that causes incorrect motor operation when using the `run_to_*_pos` command along with a non-zero `ramp_down_sp`.

@dlech has merged the fix into the `master` branch of https://github.com/ev3dev/lego-linux-drivers and it will be in the next kernel release.

kortschak, 2016-06-26:
Yes, the model moves backwards for a second. That issue may be the problem.

dlech, 2016-07-13:
Sounds like duplicate of #661

rhempel, 2016-07-17:
I got a bit further debugging this today and i think I have the solution. There is a race condition in setting tm->command. It should be done before ramping starts so that the motor gets the correct command if the ramp up time is very short.
Short ramp up times happen if the ramp_up_sp is very small or if the speed_sp is very small.
Fix has been committed here: ev3dev/lego-linux-drivers#53

kortschak, 2016-07-18:
Thank you for doing this. I will test it in the next few days.

closed 2016-07-19 by dlech

kortschak, 2016-07-21:
Confirmed fix.

rhempel, 2016-07-21:
Thks Dan! Now all I need to do is finish the Python 3 binding...

kortschak, 2016-07-21:
Somehow the kernel I built and packaged broke wireless though. I'll have a look at that more over the weekend to see what is going on.

dlech, 2016-07-21:
Which wireless? (`lsusb`)

kortschak, 2016-07-24:
```
$ lsusb 
Bus 001 Device 002: ID 0bda:8179 Realtek Semiconductor Corp. 
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

This is TL-WN725N.

dlech, 2016-07-24:
This is compiled as a separate module outside of the kernel, which is why it stopped working when you built your own kernel. https://github.com/ev3dev/rtl8188eu

kortschak, 2016-07-24:
Thanks, I figured it was probably me.