ev3dev/ev3dev #690: Possible race condition when resyncing UART sensor

bmegli, 2016-06-27:
### System information:

Problem reproducible as reported by @kortschak in https://github.com/ev3dev/ev3dev/issues/679, which was:
>
* I am using: LEGO MINDSTORMS EV3 and FATCATLAB EVB
* I installed ev3dev using this image file: ev3-ev3dev-jessie-2015-12-30.img
* My kernel version is: 4.4.13-12-ev3dev-ev3 also the release 12 on the EVB.
* My host computer is running: Linux
* I am connecting using: Wi-Fi

### About my issue:

#### What led up to the situation?

When analysing the code  for issue https://github.com/ev3dev/ev3dev/issues/679 possible race condition between user space sysfs attribute read and resyncing sensor resetting `mode_info` and waiting for new `mode_info` was identified.

As a result of race, user trying to read sensor valueX may end up with ENXIO error code (No such device or address).

The problem is non-fatal and happens when communication with sensor is failing, error message is reported to the user but it is misleading ("No such device or address").

Reproducible as @kortschak reported in https://github.com/ev3dev/ev3dev/issues/679:

>#### What exactly did you do (or not do) that was effective (or ineffective)?
Plug in FATCATLAB 9DOF to EV3 (or EVB - though slightly different outcome there due presumably to difference in cpu performance between the two).
Then navigate to device browser and click watch values.

>#### What was the outcome of this action?
The following notification by brickman for the EV3 "Error reading from file: No such device or address" and the following dmesg output:

See https://github.com/ev3dev/ev3dev/issues/679#issuecomment-228623924 for output details

>Pressing back and re-clicking watch values restores the expected brickman behaviour for the EV3, presumably because the device has stabilised after the overruns.

> Similar behaviours are seen on EVB (see #662 (comment)).

>#### What outcome did you expect instead?
Normal sensor operation and no kernel noise.

#### Explanation for race condition

I originally explained it in https://github.com/ev3dev/ev3dev/issues/679#issuecomment-228623130

Here's just a brief overview:
- uart sesnor syncs, gets mode info, ACK, sends ACK
- lego_sensor device is registered creating sysfs attributes
- if communication with device fails (e.g. `bad_message_size`) resyncing is started
- during resyncing `mode_info` is reset to default values
- during period after resetting `mode_info` and before getting new `mode_info` from UART sensor we have a race condition with userspace sysfs `valueX` attribute read

If valueX is read from sysfs during critical period mode_info `num_values` and `data_sets` are both 0 and `value_show` function in line:
https://github.com/ev3dev/lego-linux-drivers/blob/036f41ab854c3cdc21da3b6451a94fa57d2d606a/sensors/lego_sensor_class.c#L421
```
 if (index < 0 || index >= lego_sensor_get_num_values(mode_info))
        return -ENXIO;
```
returns -ENXIO

bmegli, 2016-06-27:
@kortschak I think that you are currently the only person capable of confirming that this is what happens when Brickman reports "Error reading from file: No such device or address" with Fatcatlab 9DOF. For now this is only what we are guessing. 

If you happen to build your own kernel, could you also add printk in if condition at line 421:

https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/lego_sensor_class.c#L421

Something along the line:
``` C
if (index < 0 || index >= lego_sensor_get_num_values(mode_info))
{
   printk("Value index out of bounds, returning -ENXIO \n");
   return -ENXIO;
}
```

kortschak, 2016-06-27:
Yes, I can do that, though probably not for a day or so.

kortschak, 2016-06-28:
I have rebuilt with the change above with kernel at https://github.com/ev3dev/ev3-kernel/commit/4fe7ffbf and lego drivers at https://github.com/ev3dev/lego-linux-drivers/commit/036f41a.

I now do not see the brickman error. This is what `dmesg` says:

```
[  124.397680] lego-port port0: Added new device 'in1:ev3-uart-host'
[  125.293157] Registered EV3 UART sensor line discipline. (29)
[  126.648292] lego-sensor sensor0: Registered 'ev3-uart-76' on 'in1'.
[  161.206343] ttyS1: buffer overrun
[  161.208499] ttyS1: buffer overrun
[  161.210000] ttyS1: buffer overrun
[  161.211458] ttyS1: buffer overrun
[  161.212147] ttyS1: buffer overrun
[  161.212767] ttyS1: buffer overrun
[  161.214344] ttyS1: buffer overrun
[  161.215054] ttyS1: buffer overrun
[  161.215565] ttyS1: buffer overrun
[  161.219976] ttyS1: buffer overrun
```

Am I using the wrong versions?

kortschak, 2016-06-28:
OK. Not crazy. I left it for a while and I see this:

```
[  572.764023] lego-port port0: Added new device 'in1:ev3-uart-host'
[  575.013564] lego-sensor sensor1: Registered 'ev3-uart-76' on 'in1'.
[  577.671943] lego-port port0: Removed device 'in1:ev3-uart-host'
[  577.727941] lego-sensor sensor1: Unregistered 'ev3-uart-76' on 'in1'.
[  583.025240] lego-port port0: Added new device 'in1:ev3-uart-host'
[  585.261243] lego-sensor sensor2: Registered 'ev3-uart-76' on 'in1'.
[  676.190138] ttyS1: buffer overrun
[  676.191594] ttyS1: buffer overrun
[  676.192943] ttyS1: buffer overrun
[  676.193670] ttyS1: buffer overrun
[  676.196006] ttyS1: buffer overrun
[  676.196797] ttyS1: buffer overrun
[  676.197077] ttyS1: buffer overrun
[  676.198753] ttyS1: buffer overrun
[  676.199474] ttyS1: buffer overrun
[  676.200014] ttyS1: buffer overrun
[  678.111189] Value index out of bounds, returning -ENXIO 
[  678.420658] tty ttyS1: Reconnected due to: Bad message size
```

bmegli, 2016-06-28:
Hmm, ok. Thank you @kortschak. This is important piece of knowledge.

If am getting it right it is a race condition so it sometimes happens, sometimes not, depending on timing. The brickman error seems to be direct result of ENXIO - and shows only if it happens.

So overall it seems to be...

Scenario 1:
 0.  unknown cause (but I think it's baudrate change after first sync)
 1.  bad message size byte
 2.  starting reconnect (resync)
  1. buffer overrun when circular buffer is not emptied
  2. race condition and possible -ENXIO
 3. `Bad message size` message in dmesg, printed after full resync

Scenario 2: 
 1.  buffer overrun
 2.  bad message size byte
 3.  starting reconnect (resync)
  1. buffer overrun when circular buffer is not emptied
  2. race condition and possible -ENXIO
 3. `Bad message size` message in dmesg, printed after full resync

As @dlech points out the order of messages doesn't have to exactly match order of things happening.
Confronting with your input on EVB where there was no buffer overrun I thought it was scenario 1 but I am not entirely sure.

Let's wait for @dlech opinion.

dlech, 2016-06-28:
What we really need to do here is to put a logic analyzer on one of these sensors so we can see what is actually going over the wire and compare that to what our driver is reading.

If we get a bad message size on the first data after changing baud rate, then we can't recover without re-syncing. So, we need to figure out what is going on and avoid the bad message size error in the first place.

If we get a bad message size later after we have received some data, then we already know what the message size should be and can skip one message size of data and try again a few times before resyncing just as we do for bad crc.

bmegli, 2016-06-28:
@dlech wrote
> If we get a bad message size on the first data after changing baud rate, then we can't recover without re-syncing.

Why not? Isn't it what we were doing anyway, I mean:

Currently:

``` C
while (count > 0)
	{
		/*
		 * Sometimes we get 0xFF after switching baud rates, so just
		 * ignore it.
		 */
		if ((u8)cb->buf[cb->tail] == 0xFF) {

			cb->tail++;
			if (cb->tail >= EV3_UART_BUFFER_SIZE)
				cb->tail = 0;
			count--;
			continue;
		}
		msg_size = ev3_uart_msg_size((u8)cb->buf[cb->tail]);
```

Could be replaced with (possibly with a counter limiting bad headers):

``` C
while (count > 0)
	{
		/*
		 * Sometimes after switching baud rates data is corrupted, so just
		 * ignore it.
		 */
		msg_size = ev3_uart_msg_size((u8)cb->buf[cb->tail]);

		if (msg_size > EV3_UART_MAX_MESSAGE_SIZE) {

			cb->tail++;
			if (cb->tail >= EV3_UART_BUFFER_SIZE)
				cb->tail = 0;
			count--;
			continue;
		}
```

But it's true that it would be best to know what is really going on.

dlech, 2016-07-01:
So how do you know how many bytes to skip? Did we receive one extra bad byte? Did we receive 1/2 of a message? Did we receive a full length message, but with corrupted data? This is why we need to compare what the driver is receiving vs. what is actually being transmitted from the sensor.

bmegli, 2016-07-07:
You are probably right, as usual. I have to think it over.

Interestingly I am also getting occasional ENXIO when reading from I2C device using direct attribute (tight loop blocking only on i2c read and reads of motor positions/speeds). I am always getting it after a few minutes

I will fill up new issue when I gather more information.