ev3dev/ev3dev #692: Servo state stuck in "ramping" after short, quick movement

Redoubts, 2016-06-30:
### System information:

* I am using:  LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3-ev3dev-jessie-2015-12-30.img

* My kernel version is: Linux ev3dev 4.4.13-12-ev3dev-ev3 # 1 PREEMPT Tue Jun 14 20:26:05 CDT 2016 armv5tejl GNU/Linux

* My host computer is running: OS X

* I am connecting using: Bluetooth

### About my issue:

If the ramp speed is too long, relative to the degrees moved, the servo gets stuck in state `ramping`. 

Example using the ev3 large servo motor: 
I set 
* `stop_action` to `brake`
* `position` to `0`
* `position_sp` to `10`
* `ramp_*_sp` to `500`
* `speed_sp` to `480`

Then do `run-to-rel-pos`. 

The servo moves as desired, but if I check `state` I get `ramping` indefinitely, until I do something else. If I cut down the sp to something like 50, then the servo correctly reports an empty string when its done.

Guess: Servo finished moving before it hit the speed setpoint, which means it never finished ramping up to the set point.

This is bad because I'm writing a script that needs to move on after it's determined the servo finished moving.

rhempel, 2016-06-30:
I have what I believe is a good fix in the current `lego-linux-drivers` repo, but it has not made its way to the kernel yet. I will try with your test data to see if the new driver works correctly.

Just tried it, the latest driver in kernel source works just fine - you'll have to wait or maybe @dlech can build an updated kernel for release:-)

dlech, 2016-07-01:
So, duplicate of #661?

rhempel, 2016-07-04:
Yep - closing with this note that it's a Dup=uplicate of #661

closed 2016-07-04 by rhempel