ev3dev/ev3dev #695: Serial port communciation

mlvandenbos, 2016-07-04:
Is it possible to use a sensor port to have serial communication? I don't want to connect a sensor but to connect a ZigBee interface and use that as a 'wireless' serial port.

As a test I connected a serial cable to sensor port nr 2.

I did
`echo "other-uart" > /sys/class/lego-port/port1/mode `

to set the port to be 'other-uart' (the port numberingis port0/port1 etc so I assumed that port1 -> sensor port 2)

I was then able to send some test data with

`echo -ne "hello world" > /dev/ttyS0`

This data arrived at the other end of the cable.

The problem now is sending data back. I don't seem to get any data in the EV3. What can I do to test?
I tried with

`cat /dev/ttyS0`

If I'm able to send/receive data this way, I will go on and write a c++ application to handle the data. The console part is only to confirm the setup could work.

dlech, 2016-07-04:
Yes, this should work and it looks like you are doing everything correctly. I think you just need to add a newline to the end of the data you are sending back or change the tty settings.

Tips:

* Don't assume that `port1` is sensor port 2. It usually is, but may not always be that way. Check the `address` attribute to be sure.
* You can also use the Device Browser on Brickman to change the mode of the port.
* There are symlinks that use the port names, so you can use `/dev/tty_in2` instead of `/dev/ttyS0`.
* You can use `stty --file=/dev/tty_in2` to view the current tty settings.
* Use `stty --help` to see how to change settings from the command line.
* In your C++ program, you can use [termios](https://manpages.debian.org/cgi-bin/man.cgi?query=termios) functions to configure the tty.

mlvandenbos, 2016-07-04:
Thanks for the quick reply. 

The newline did the trick ;-)

Thanks for the extra information, that will be helpfull along the way.

closed 2019-03-25 by dlech