ev3dev/ev3dev #696: PiStorms: motor jitters if stop_action is hold

dwalton76, 2016-07-10:
### System information:

* I am using: Rapsberry Pi Model 2B

* I installed ev3dev using this image file: rpi2-ev3dev-jessie-2015-12-30.img.xz

* My kernel version is: 4.4.13-12-ev3dev-rpi2

* My host computer is running: Ubuntu 16.04

* I am connecting using: WiFi

### About my issue:

When I run the following the motor stops but then jitters back and forth as shown in https://www.youtube.com/watch?v=NjeqUuCAIHc
```
echo reset > /sys/class/tacho-motor/motor1/command
echo hold > /sys/class/tacho-motor/motor1/stop_action
echo 400 > /sys/class/tacho-motor/motor1/speed_sp
echo 1080 > /sys/class/tacho-motor/motor1/position_sp
echo run-to-abs-pos > /sys/class/tacho-motor/motor1/command
```

The trigger is setting the stop_action to hold, this is what causes the motor to shake. 'brake' and 'coast' are both fine.

@dlech mentioned that I might have an older pistorms.  Can someone else with a pistorms see if they hit this issue?

dlech, 2016-07-10:
I had the same issue with my original PiStorms. When I contacted mindsensors.com about it, they exchanged it for a new PiStorms-v2. They mentioned that they had a hardware problem on very early model PiStorms.

I can't reproduce the issue on the v2. There is probably not a way to do anything about this in the ev3dev kernel drivers. I would suggest contacting mindsensors.com. It may be possible that a firmware upgrade may fix the problem or they may exchange it like they did for me.

dwalton76, 2016-07-10:
Ack thanks David.  I'll ping them and see if I can get a replacement....will re-open this ticket if I still see this on newer hardware.

dwalton76, 2016-08-21:
mindsensors sent me a new pistorms and it has the same issue

dwalton76, 2016-08-21:
I am running:
```
robot@raspberrypi2[utils]# uname -a
Linux raspberrypi2 4.4.17-14-ev3dev-rpi2 #1 SMP Mon Aug 15 15:42:55 CDT 2016 armv7l GNU/Linux
robot@raspberrypi2[utils]# 
```

dlech, 2016-08-21:
I can't repeat the problem using the script provided in the original post. It is working correctly for me. Does it work with the official PiStorms software?

dwalton76, 2016-08-21:
I'll give it a try

dwalton76, 2016-08-27:
I tried PiStorms software and see the same issue.  Either I happened to get another bad PiStorms board or the PiStorms is just a piece of junk :(  Based on the amount of activity on their forum I'm guessing they haven't sold many of these.

I posted on their forum about this issue:
http://www.mindsensors.com/forum/topic/205_pistorms-motor-issues

closed 2016-08-27 by dwalton76

dlech, 2016-08-27:
The new video you linked from the forum post shows a different behavior from the video in your original post here. In the original video, the motor kept rotating past the point where is was supposed to stop. In the new video, the motor actually stops. I would say that the new hardware has fixed the original issue. The new behavior may still be undesirable, but it is different.

dwalton76, 2016-08-28:
It feels like the same issue, just less power is going to the motor with PiStorms SW and it isn't enough to make the motor actually move.  In both cases the trigger is using stop_action hold.