ev3dev/ev3dev #699: ev3dev using multiple sensors with python

jamesc601, 2016-07-12:
Hi.

I am completely new to ev3dev and I am currently looking in to a potential project. This may be done with either lego mindstorm or a raspberry pi. I plan on using multiple infrared sensors in this project. However, I can not seem to find a way to check what a specific sensor senses as all example code I have found seems to only use one sensor and does not identify it by it's port but just as a sensor. Is there a way to do this?

Thanks in advance

dlech, 2016-07-12:
You can use the `address` attribute to figure out what port a sensor is connected to.

WasabiFan, 2016-07-12:
Assuming that you are using the library from <https://github.com/rhempel/ev3dev-lang-python/> (which is the only up-to-date one as far as I know), something like this should work:

```
m = ev3.LargeMotor('outA')
m.run_timed(time_sp=3000, duty_cycle_sp=75)
```

You can find other examples of usage in the readme, and their docs should include info on the constructor parameters you need.

WasabiFan, 2016-07-12:
That example code is for a motor, but sensors should work similarly too; the sensor ports are called `inA`, `inB`, etc.

ec-m, 2017-10-25:
Actually, the sensor ports are called `in2`, `in3` etc. With this you can use multiple ultrasonic sensors in the following way:

`us_1 = UltrasonicSensor('in3')`
`us_2 = UltrasonicSensor('in4')`
`assert us_1.connected, "Connect a single US sensor to any sensor port"`
`assert us_2.connected, "Connect a single US sensor to any sensor port"`
`us_1.mode='US-DIST-CM'`
`us_2.mode='US-DIST-CM'`
`units = us_1.units`

`distance_1 = us_1.value() / 10  # convert mm to cm`
`print(str(distance_1) + " " + units)`
`distance_2 = us_2.value() / 10`
`print(str(distance_2) + " " + units)`

ddemidov, 2017-10-25:
In stretch the port names have been changed to `ev3-ports:in<n>`, `ev3-ports:out<c>`. So it is best to use the predefined constants `INPUT_<n>`, `OUTPUT_<c>`.

ec-m, 2017-10-25:
@ddemidov Thank you for this information. Then the correct way would be

`us_1 = UltrasonicSensor(INPUT_3)`
`us_2 = UltrasonicSensor(INPUT_4)`

closed 2019-03-25 by dlech