ev3dev/ev3dev #702: SSH broken in new images (nightly)

WasabiFan, 2016-07-18:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3-nightly-2016-07-13-bootroot.img.xz

* My kernel version is: Not accessible

* My host computer is running: Windows and Ubuntu

* I am connecting using: Wi-Fi

### About my issue:

After installing the nightly image, I am unable to SSH into the brick (using a username/password).

```
$ ssh robot@ev3dev.local -v
OpenSSH_6.9p1 Ubuntu-2ubuntu0.2, OpenSSL 1.0.2d 9 Jul 2015
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 19: Applying options for *
debug1: Connecting to ev3dev.local [192.168.1.21] port 22.
debug1: Connection established.
debug1: identity file /home/wasabi/.ssh/id_rsa type 1
debug1: key_load_public: No such file or directory
debug1: identity file /home/wasabi/.ssh/id_rsa-cert type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/wasabi/.ssh/id_dsa type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/wasabi/.ssh/id_dsa-cert type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/wasabi/.ssh/id_ecdsa type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/wasabi/.ssh/id_ecdsa-cert type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/wasabi/.ssh/id_ed25519 type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/wasabi/.ssh/id_ed25519-cert type -1
debug1: Enabling compatibility mode for protocol 2.0
debug1: Local version string SSH-2.0-OpenSSH_6.9p1 Ubuntu-2ubuntu0.2
debug1: Remote protocol version 2.0, remote software version OpenSSH_6.7p1 Debian-5+deb8u2
debug1: match: OpenSSH_6.7p1 Debian-5+deb8u2 pat OpenSSH* compat 0x04000000
debug1: Authenticating to ev3dev.local:22 as 'robot'
debug1: SSH2_MSG_KEXINIT sent
Connection reset by 192.168.1.21
```

dlech, 2016-07-18:
I think I have this fixed. Please try the nightly when it is done building.

WasabiFan, 2016-07-18:
Whoops, wrong button 😕

WasabiFan, 2016-07-19:
I have just imaged by SD card with the nightly from today and the SSH issue still persists. The verbose logging on the client side is the same as above.

dlech, 2016-07-19:
This is a head scratcher. The CI build ends up with different permissions on these files than my local trusty VM. Before f845764, the permissions in the CI image were 666. After the commit, they are 660. They should be 600, which is what I get when I build an image locally.

WasabiFan, 2016-07-19:
Hmmmm... Maybe something odd about Travis' caching? Possibly a special case for one of the packages brickstrap uses? Seems like a longshot, but I suppose it's possible.

WasabiFan, 2016-07-19:
Travis publishes Docker images of their build environment; testing might be easier if you ran that locally instead of waiting on their slower infrastructure.

dlech, 2016-07-19:
We're not using docker on travis, we are using the trusty full VM.

WasabiFan, 2016-07-19:
Oh, I see... that's unfortunate. Let's just hope that it doesn't take many cycles of trial-and-error before you figure it out 😆

dlech, 2016-07-19:
your wish has come true.

WasabiFan, 2016-07-19:
Ah, nice; I'll try out the build now.

WasabiFan, 2016-07-19:
Just tested it; I can confirm that SSH now works in the nightlies.

dlech, 2016-07-19:
Fix released in brickstrap 0.5.1

closed 2016-07-19 by dlech