ev3dev/ev3dev #703: Runing C program on the EV3

volansky, 2016-07-24:
Hey,
I'm trying to run a simple program in C on the ev3dev but really don't understand how to.
I got the ev3dev runing on my EV3 but I can't get seem to understand how exactly can I run the program on the EV3, I read some of the tutorials/question but for someone who's new to all this and never used Linux it's really confusing and seems much more complicated than it should be as all I want to do is just upload a program and run it.
Is there an easy way to run a written program in C on the ev3? something like moving the .C file to the EV3 using filezilla and then run it from the brick? and if not is there some documentation on how exactly one can acheive that?
Any help will be greatly appreciated.
Thank you.

theZiz, 2016-07-24:
Well, C programs need to be compiled before running. One possibility is of course to compile it directly on the EV3. Are you familiar with Makefiles? In that case you could always call `make` before running your compiled program itself.

However I guess the best idea for your is first of all to work with linux a bit. So maybe install a Linux Mint into a VM (like Virtualbox) and read a bit about gcc, Makefiles and litte bashscripts before switching to ev3dev.

Afaik there is no easy "Click a button and it runs" way for now.

volansky, 2016-07-24:
Thanks for the quick response, my end goal is that the EV3 will run my program without any connection to a computer ( I want to take an EV3 unit insert the SD card and start the program without using any PC) so I assume that the program will have to be compiled on the EV3 brick right? which really narrow my options to the first one.

volansky, 2016-07-24:
I created a make file and copied it to the EV3 using filezilla but when trying to run it the brick writes that the file isn't an executable or a folder. am I doing it wrong? isn't the makefile suppose to compile and run the other files?

moriarty, 2016-07-24:
Did you try to execute the makefile from the brick UI?

You'll need to ssh into the ev3 and run the make command, which will follow the instructions in the makefile you've copied and compile your project.

If the project is large and doesn't compile on the EV3 or takes too long, you should follow the advice of @theZiz and look into Linux, and then look into using brickstrap.

http://www.ev3dev.org/docs/tutorials/using-brickstrap-to-cross-compile/

volansky, 2016-07-24:
So if I understand you correctly, what you say is that there is no way for the EV3 to run a C program on its own meaning there most be a pc involved (to SSH and run the make command or by compiling on it using the brickstrip), is that right?
I'm sorry if i'm being repetitive but It's crucial for my project, if I have to connect the EV3 to a pc each time I want to run the program, using ev3dev isn't the solution I was looking for (sadly).

WasabiFan, 2016-07-24:
_Side note:_ In the future, we need to make sure that we aren't confusing beginners. We need to be explaining these things in enough detail to get them going and then preferably document it for future users.

---
@volansky:

Firstly, it sounds like you have already set networking up and are using Filezilla. Great! That is the first barrier to entry, and you seem to have figured that one out pretty easily.

To understand what you need to do, you'll first need to understand the way that C programs are run. C is a "compiled" language; that means that the code that you write must be converted into a different kind of code that a computer (in this case the EV3) understands. This is the job of the "compiler": it is a program that takes your C code and converts it into something that can be run. If you are familiar with Windows, the final product is the equivalent to a `.exe` file. On ev3dev, you will need to run the compiler over SSH to create the executable file, but once you have that you can run it whenever you want from Brickman.

So, how do you compile your code? There are many ways to do it, but the one that I would recommend to you is called a "makefile." A makefile is a file that gives instructions on how your program should be compiled; don't worry, it's actually pretty simple.

Create a file and call it `Makefile` (with no extension) in the directory with your code. Copy the following contents into it:

```makefile
all: main.o
    gcc main.o -o bin
main.o: main.c
    gcc -I . -c main.c
clean:
    rm -rf *.o
    rm bin
```

This code assumes that your C file is called `main.c`. If it is called something else, either replace the mentions of `main.c` or rename your file. Also, you will need to make sure that the indentation is tabs and not just 4 spaces (which I believe GitHub uses). You can delete the spaces and then hit <kbd>Tab</kbd> to replace them.

Now, from an SSH session, run `make`. If it is successful, the output should look like this:

```
gcc -I . -c main.c
gcc main.o -o bin
```

Now, assuming it succeeded, you should have a file called `bin` (you can run `ls` to see what's in the current directory). You can run your new file by typing `./bin`. You can also run that file from Brickman.

If anything in those steps failed, please give us the full error output.

*Some warnings:*
- If you want to use the motors, sensors, etc. of the EV3, you will need to either find an existing C library (not C++) or manually read from and write to the files that let you control devices. I am not aware of any C libraries that are still maintained, so you'll need to do some research (our website's "libraries" page is a good start). We can help you with modifying the makefile above to integrate a library.
- The version of the kernel (the core part of ev3dev, including the drivers that control motors, sensors, etc.) that comes in the December image release is pretty old. Depending on the version of ev3dev that the library you choose supports, you may need to upgrade your kernel. You can upgrade to the latest as follows:

    ```
    sudo apt-get update
    sudo apt-get upgrade
    sudo apt-get dist-upgrade
    ```
    Note that it will take a long time, possibly multiple hours. The EV3 must be connected to the internet for this to work.

rhempel, 2016-07-24:
@WasabiFan - I took a break writing up my crappy notes below and in the meantime you put together a really good set of instructions that I think deserve a page on the ev3dev website. Please consider putting this on the site so that when we get questions like this in the future we can simply point to your notes :-)

@volansky - your mental model of what is happening is not quite accutrate. You can compile C programs on the brick, but you need to install a package called `build-essentials`. This will take a long time to download and install.

You can also compile programs on your host machine (pc) but then you need a cross compiling environment.

The easiest thing for you to do (if you are an inexperienced Linux user or programmer) is to install the compiler on the brick. Compiles will be slow but there's a lot less complexity.

Either way, your compiled C programs end up on the brick, and as long as the execute bit is set, you can run them from the File Browser that is part of Brickman.

WasabiFan, 2016-07-24:
@rhempel:

> I took a break writing up my crappy notes below and in the meantime you put together a really good set of instructions that I think deserve a page on the ev3dev website. Please consider putting this [...]

I think I'll do that. I've wanted to get beginner documentation started for a while, so I guess this would be a good starting point. And by the way, your notes aren't crappy 😆 they just are at a higher level.

> but you need to install a package called build-essentials

Crap. I forgot to check if `build-essential` was installed by default, and it looks like it isn't. @volansky I added a note above to install that; make sure you do.

---

@volansky:

> it's really confusing and seems much more complicated than it should be as all I want to do is just upload a program and run it.

Unfortunately, this is the nature of both C and Linux, so I am not particularly surprised. We have wanted to write beginner tutorials for a while, but haven't actually gotten around to doing it yet. We still consider ev3dev a "beta" that is targeted at intermediate and above skill levels, but we plan to expand our documentation enough to suit beginners. I am often the resident "beginner advocate" in that I am the one who pushes for beginner docs and features. I guess those will probably start with my notes from above.

Also, if you have questions that would be easier to answer in chat, you can join our GItter chatroom [here](https://gitter.im/ev3dev/chat).

volansky, 2016-07-24:
I think I'll explain what I'm trying to do so you guys could understand me better.
I work in a small startup, we create a physical programming experience for children using special  blocks on which each block represent a different kind of a command and connecting the blocks will form a program. 
our product comes with our own robot but we'd like to enable owners of EV3 (and NXT) to be able to connect to the blocks using their bricks so they wont have to buy the robot, to do so we need to find the easiest way to get the EV3 to run our robots program (which is written in C) and by the easiest I mean that the user will have to take minimal steps to get it running, ideally all they will have to do is download a folder to the EV3 and run the program from the brick's UI, anything more than that and it will be too complicated, also there is no need for more than one compilation as the code won't be changed.

Is that possible? and if so does it matter which way I choose?
also is it possible to get a compiled file and copy it to a different brick and run it from there? 

Thank you very much for all the help, you guys rock!

WasabiFan, 2016-07-24:
Ah, I see; that use case is a bit different. To make sure I understand, am I correct in that you don't have any of your code running on the user's PC? If so, I see two options:

1. Pre-compile the code and just give the user a binary. You can compile the code on your own brick and then give it to others, or if you are willing to get into some Linux work, you can try Brickstrap as mentioned above.
2. Write a script that compiles your code. If you are familiar with Python, use that; shell scripts ("bash") would work as well. What your script would need to do is install a C compiler (`build-essential`) and then compile your code and run it. It could be as simple as this (bash):

    ```bash
    #!/bin/bash

    # Install a compiler if it isn't installed
    dpkg-query -l build-essential
    ret=$?
    if [[ ret != 0 ]]; then
        sudo apt-get install build-essential
    fi

    # Compile the code if it hasn't been compiled already
    if [ ! -f bin ]; then
        gcc -I . -c main.c
        gcc main.o -o bin
    fi

    # Run the binary
    ./bin
    ```
    This is pretty ugly (and spews a bunch of unintelligible output) but it works as far as I can tell. This script can be run from the brick display.

volansky, 2016-07-24:
You are correct.  
The first option sounds perfect as I'd rather save the user the compilation process so giving him the compiled program will be perfect for my needs.
So what I need to do is get my program compiled (with one of the options mentioned above) then put the folder on the ev3dev SD and all the user will have to do is inseret the sd and run the executable from the brick UI?

WasabiFan, 2016-07-24:
> So what I need to do is get my program compiled (with one of the options mentioned above) then put the folder on the ev3dev SD and all the user will have to do is inseret the sd and run the executable from the brick UI?

Not quite. The SD card is just for system files, so you can't easily put your program files on it. You will probably want to put your program files on a USB flash drive and plug it into the USB port, or boot the SD card on your own EV3 beforehand and save the files there before giving the card to a user. There may be other options that I am not considering; @dlech and @rhempel, are there other options there? I have opened #705 which would address this.

dlech, 2016-07-24:
I don't think I understand the entire big picture here. Is the non-robot part of your product a programing environment like the official EV3 software/Scratch/OpenRoberta/etc.?

What exactly does this C program do? Does is run other programs? Does it allow remote control of the EV3?

How often would this C program need to be updated by end users?

WasabiFan, 2016-07-24:
> I don't think I understand the entire big picture here. Is the non-robot part of your product a programing environment like the official EV3 software/Scratch/OpenRoberta/etc.?

I missed that too at first. @volansky said (emphasis mine):

"I work in a small startup, we create a **physical** programming experience for children using special blocks "

I think that is the answer you're looking for.

dlech, 2016-07-24:
perhaps like this? http://www.wired.co.uk/article/primo

dlech, 2016-07-24:
I still have the same questions though.

volansky, 2016-07-25:
It's more like this project - https://www.youtube.com/watch?v=AuRTS35ouTs

> What exactly does this C program do? Does is run other programs? Does it allow remote control of the EV3?

The C program (on the EV3) receives a Bluetooth stream of a set of commands from another controller (the blocks), after receiving it follows the commands and send back a message saying which command is being performed.

> How often would this C program need to be updated by end users?

Around 6 months.

dlech, 2016-07-25:
If the program is more than a single file, I would recommend packaging it up in a debian package.

And since this is more like a service for a programming platform, we could even integrate it into brickman like we have done with OpenRoberta.

volansky, 2016-07-26:
That will be great, when I'll get to that point I'll mentioned that option, untill then there's a lot I need to learn to do.

dlech, 2016-08-05:
FYI, the new "official" way to compile C/C++ programs for ev3dev: http://www.ev3dev.org/docs/tutorials/using-docker-to-cross-compile/

closed 2019-03-25 by dlech