ev3dev/ev3dev #705: Allow file transfer on system SD card

WasabiFan, 2016-07-24:
Would it be possible to add a partition in the default image that would allow the user to transfer files from their PC? Currently, you can't really use the SD card for file storage (and it is especially difficult on Windows). I am imagining something small -- maybe even only 100 MB or so by default -- that would automatically be mounted on boot and could be used for log transfers and program files.

rhempel, 2016-07-24:
I think we should rather focus on getting Windows users (and others) to use FileZilla or other methods of transferring files. Windows can sometimes ask the user to format a "drive" if all the partitions are not recognized. I'd prefer to have the SD card stay in the brick if at all possible.

WasabiFan, 2016-07-24:
I agree that we should promote the networked options and put in effort to improve them, but for people who are having networking issues (whether that is because of a bug, unavailability, or the user being unable to) the built-in option to use the SD card would be helpful. The alternative is finding a USB hub, keyboard and flash drive and copying via the terminal, which could be difficult, especially for newbies who would otherwise just run from Brickman.

dlech, 2016-07-24:
There is already a fat partition that can be used for this. It mounts at `/boot/flash`. But I'm with @rhempel, I'm not really a fan of doing things that way. It's unfortunate that network connectivity can be such a pain though.

There is the possibility creating an extra partition on the SD card and using a usb gadget driver to provide a mass storage driver. On catch to this is that you can't have the mass storage driver running at the same time that you have the partition mounted on the EV3. So, to make this work, we would have to bring back the USB menu in brickman. To transfer a file, the user would have to select the mass storage option in the USB menu on the EV3. Then transfer the file on the desktop machine. Then select something other than mass storage (or unplug the USB) in order to use the files on the EV3. Not ideal either, but much better than removing the SD card.

markyi370, 2017-05-16:
@dlech maybe you guys should implement MTP, like Android did back in 4.1.

dlech, 2017-10-28:
Is this still useful now that we have the VS Code extension?

WasabiFan, 2017-10-28:
Once there's a "view remote file" command (I think you've tried out some options for that but it isn't in the shipping version AFAIK) I would say this is resolved.

dlech, 2017-10-28:
There is an Upload command. Is that good enough? 😄

WasabiFan, 2017-10-28:
The upload command operates on the files in your workspace directory, correct? My recollection of the scenario here was that I ran scripts from the command line and then wanted to retrieve the logs. If the "upload" command works on arbitrary files I'm happy with that.

dlech, 2017-10-28:
The upload command works on remote files listed in the ev3dev browser. So yes, you can currently upload log files created on the EV3 if you want to.

dlech, 2017-10-28:
Also, VS Code has plans to add built-in remote file systems, so in the somewhat near future, we should be able to open a remote folder on the EV3 just as you would a local folder on your hard drive.

WasabiFan, 2017-10-28:
It looks like the upload command offers a "save" dialog to let you grab arbitrary files, so I think it satisfies this scenario pretty well.

closed 2017-10-28 by WasabiFan