ev3dev/ev3dev #707: khidpd high cpu usage

meehien, 2016-07-25:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3-nightly-2016-07-24-bootroot.img

* My kernel version is: v4.4.15-13-ev3dev-ev3

* My host computer is running: Archlinux

* I am connecting using: wifi

### About my issue:

Hey guys,

Sorry to be a nuisance, but seems to be a persisting problem with Bluetooth connections between EV3 and PS4 controllers (and maybe other Bluetooth input devices). I was unable to properly tell what was causing it before (as I had done numerous modifications to my running image), but now the nightly-es  make everything much more easy to install and test.

So on a vanilla installation of the latest nightly (2016-Jul-24, kernel: v4.4.15-13-ev3dev-ev3), when I connect the Bluetooth PS4 controller the khidpd_054c05c4 process starts using about 20% of CPU and makes everything lag (Brickman, SSH, etc.) If the controller is not connected then everything works perfectly. I was unable to find much information about the khidpd process and how this has changed from kernel 3.14 to 4.4.x, nut to me this seems to be the culprit.

dlech, 2016-07-25:
So there was not a noticeable lag with the 3.14 kernel? I don't have a PS4 controller to test, so I can't be much help. Google doesn't turn up anything either.

I tried a bluetooth keyboard and `khidpd` is the very last process listed in `top`.

I suppose you could try building your own kernel. There are a couple of PS3/4 controller related [commits](https://github.com/ev3dev/ev3-kernel/commits/ev3dev-jessie) we have applied on top of the mainline kernel that could affect it.

If it was working OK in earlier kernels but not now, the only thing to do really is to do a git bisect and compile and test kernels until you find when the problem started.

meehien, 2016-07-31:
I have found a fix for this. Line 1882 in drivers/hid/hid-sony.c needs to buf[1] = 0xB0 (as it was in kernel 3.16) instead of buf[1] = 0x80. This value controls DS4 report frequency and it seems that, even though 0x80 might be good for more powerful hardware, it's a total CPU hog on the EV3 brick.

Thanks to Frank Praznik for pointing me on the right direction.

dlech, 2016-07-31:
Will you make a pull request to https://github.com/ev3dev/ev3-kernel?

meehien, 2016-07-31:
given it's such a small change I wasn't planning to. But I can if that is what you prefer.

dlech, 2016-07-31:
One less thing for me to do. :smiley:

meehien, 2016-08-01:
> One less thing for me to do. 😃
 
wait, does this mean the chance will be included or not? cause i definitely want the change to be included. 😃

WasabiFan, 2016-08-01:
He just means that making the change himself takes time on his part. He's already busy with other stuff, so anything that others can do (like making that change and opening a PR) is helpful to lessen his workload.

meehien, 2016-08-01:
Cool. I totally understand that. In that case I'll do a PR.

dlech, 2016-09-04:
Fix released in `14-ev3dev` kernel. Please test an repoen if this is still a problem.

closed 2016-09-04 by dlech