ev3dev/ev3dev #708: Creating SSH connection with the ev3dev using Ubuntu on VirtualBox

volansky, 2016-07-26:
I'm trying to create an SSH connection between the EV3 and my virtual machine but no luck.
Went through the "Connecting to the Internet via USB tutorial, all went well up to the point I need to call the brick using `ssh robot@ev3dev.local` which returns:

> ssh: Could not resolve hostname ev3dev.local: name or service not known

I tried pinging the device using its IP address and the hostname but nothing came back.

Using lsusb returns:
> Bus 001 Device 003: ID 0694:0005 Lego Group

The wired connection on the brick says >State: connected

I'm using the latest versions of both the ev3dev and the VirtualMachine/Ubuntu

dlech, 2016-07-26:
Saying "latest" version is not helpful. Please say the exact versions.

The fact that the brick says "connected" rather than "online" tells me that you have not configured the connection correctly on the host computer side and that it is probably using a local link address (169.254.x.x.).

volansky, 2016-07-26:
Sorry about that I wasn't thinking, I edited my question.

dlech, 2016-07-26:
Please try the nightly image.

[ [image] ](https://bintray.com/ev3dev/nightly/ev3dev-jessie/_latestVersion)

volansky, 2016-07-26:
I tried nightly and checked the IPv4 address on the host-only configuration (192.168.137.3) and can now ping the ip address but still can't ssh. 
The connection on the brick is still set to connected.

dlech, 2016-07-26:
What is the IP address of your Ubuntu VM?

volansky, 2016-07-26:
Im not sure if im doing it right so I attached this capture

[image]

dlech, 2016-07-26:
I would expect to see an entry with `usb0` there. This tells me that USB is not connected or that the kernel driver in Ubuntu 16.04 no longer works.

dlech, 2016-07-26:
or the EV3 is messed up because you are connecting it to windows first and then trying to us it in a virtual machine.

volansky, 2016-07-26:
Any idea on how to approach this? (except trying an earlier version of ubuntu)

dlech, 2016-07-26:
Can you connect on Windows?

volansky, 2016-07-27:
Yes, using putty/filezilla I was able to connect.
(Wired state: connected)

dlech, 2016-07-27:
OK, so we know that the problem is not with ev3dev.

Are you using the USB forwarding feature of VirtualBox to forward the USB connection to the Ubuntu VM?

volansky, 2016-07-27:
I wasn't, just set it up but the results are the same.

(For future users I used this tutorial to set the usb forwarding feature - http://www.arvydas.co.uk/2011/02/forwarding-usb-devices-on-oracle-virtualbox/)

WasabiFan, 2016-07-27:
@volansky What exactly have you done so far to configure networking? i.e., which guides have you followed? If you were unaware that you needed to configure USB forwarding, my guess is that there was another step that was missed along with it.

volansky, 2016-07-28:
That's probably correct.
I download the ssh and upgraded it, opened the port in firewall, configured a host only adapter (with the EV's ip), configured the USB forwarding, set the wired connection in the EV3 to linux (I think that's all).

I wasn't following any particular tutorial as I couldn't find anything comprehensive that goes through the whole process.

WasabiFan, 2016-07-28:
Did you follow any of the tutorials on [this page](http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/)?

Depending on what you are trying to do, you might have the best luck with trying to configure USB network sharing on the Windows PC and then setting the VirtualBox network configuration to be bridged with the host PC. That way, both the Windows PC and the Ubuntu VM can access the EV3. If you do it this way, you will want to disable USB forwarding in the VM.

Otherwise, you should follow the Ubuntu section with USB forwarding in VirtualBox on.

volansky, 2016-07-31:
Yea following that tutorial was the first thing I did and when I wasn't able to connect I opened this thread.
Today I was able to ssh the brick, not sure how because nothing has changed since last time I tried  but I guess I'll leave it like that.

Thanks for all the help.

markyi370, 2017-05-16:
Please run `lsusb` in the VM and post the output.