ev3dev/ev3dev #710: EVB large motor rotates too far when speed >= 400

dwalton76, 2016-07-30:
### System information:

* I am using: Fatcatlab EVB

* I installed ev3dev using this image file: evb-ev3dev-jessie-2016-01-25

* My kernel version is: 4.4.15-ti-rt-r37-13-ev3dev-bb.org

* My host computer is running: Linux

* I am connecting using: wifi

### About my issue:

When I run the following the motor (large) over rotates. It says that it stopped at 1080 but it went further than that...maybe another 60 degrees (eyeball guess).

robot@beaglebone[motor1]# echo reset > command
robot@beaglebone[motor1]# echo 1080 > position_sp 
robot@beaglebone[motor1]# echo 400 > speed_sp
robot@beaglebone[motor1]# echo hold > stop_action
robot@beaglebone[motor1]# echo run-to-rel-pos > command
robot@beaglebone[motor1]# cat position
1080
robot@beaglebone[motor1]# 

Neither brake vs hold or run-to-rel-pos vs run-to-abs-pos matter, I can reproduce it with all of them. If I drop the speed down to 300 or lower then it works just fine.  When using a speed a 400 the motor seems to jump right at the beginning and then continues moving at the normal speed.

dlech, 2016-07-31:
It is expected that the motor would not stop an an exact position with `brake`. It just slows the motor down quicker that `coast`.

If the motor does not stop at an exact position with `hold` and actively maintain that position, then that is a bug.

dwalton76, 2016-08-01:
I was testing with `hold` initially...looks like a bug

dlech, 2016-08-03:
I have just noticed that the EVB inverts the signal on pin 6 where the EV3 does not. I think changing `0` to `GPIO_ACTIVE_LOW` for these pins in the device tree might fix the problem.

https://github.com/ev3dev/bb.org-kernel/blob/ev3dev-jessie/arch/arm/boot/dts/am335x-bone-evb.dtsi#L152

I have attached an updated .dtb for testing. Backup and replace the file in `/usr/lib/$(uname -r)/`

[am335x-boneblack-evb.dtb.zip](https://github.com/ev3dev/ev3dev/files/400241/am335x-boneblack-evb.dtb.zip)

dwalton76, 2016-08-06:
No joy :(

dlech, 2016-08-06:
Is the problem reproducible on the EV3?

dwalton76, 2016-08-06:
Na it works fine on EV3

dlech, 2016-08-23:
I've finally had a chance to look at this. At a speed of `200`, things seems to work OK, but at high speeds I am seeing over-rotation as well. I have a theory that we are missing interrupts on the gpios that measure rotation of the motor. I'm not sure why/how we could be missing them though since this is a more powerful processor than the EV3 and the same driver works fine on the EV3.

dwalton76, 2016-09-14:
@dlech does it work for you with speed of 400 with the pin commit?  I had tried a few weeks ago and didn't have any luck but I am guessing I botched something with loading the .dtb

dlech, 2016-09-14:
No, I am able to reproduce the issue. It seems to be missing interrupts for some reason. In the work I am doing for `lms2012-compat`, I ported the motor drivers from the official LEGO firmware/official EVB firmware and it seems to be working, so apparently we are doing something wrong in the ev3dev drivers, but there is nothing obviously wrong that I can see.

I'm sure someone will figure it out sooner or later.

dwalton76, 2016-11-27:
FYI I upgraded my EVB to 4.4.31-ti-rt-r67-17-ev3dev-bb.org and am no longer seeing this issue

dlech, 2016-11-28:
Well, it would be nice to know what actually changed that fixed it, but I guess we will close the issue.

closed 2016-11-28 by dlech

dlech, 2017-01-24:
FWIW, this issue still exists in kernel 4.4.31-ti-rt-r67-17-ev3dev-bb.org. It just requires a higher speed to observe the problem. Tested using `echo 1200 > speed_sp` with medium motor on EVB.

Leaving the issue closed since it is a duplicate of #795, which seems to have more useful information.