ev3dev/ev3dev #711: BT 4.0 BLE support in new EV3

JorgePe, 2016-07-31:
This is related to #168 and #521.
I apologize for the lack of solid information but I have a feeling that the new EV3 BT does support BLE.

I finally got my new EV3, so I decided to try BLE. As already said, it shows no lestates or LE related features. But the fact that it reports HCI 4.0 LMP 4.0 intrigues me and after some googling I found
that running this commands seem to enable LE features on the Texas hci0 device

```
sudo hcitool cmd 0x3f 0x015B 0x1 0x0
< HCI Command: ogf 0x3f, ocf 0x015b, plen 2
  01 00 
> HCI Event: 0x0e plen 4
  01 5B FD 1F

sudo hcitool cmd 0x3f 0x01DD 0x1
< HCI Command: ogf 0x3f, ocf 0x01dd, plen 1
  01 
> HCI Event: 0x0e plen 4
  01 DD FD 00

sudo hciconfig hci0 lestates
Supported link layer states:
	YES Non-connectable Advertising State
	YES Scannable Advertising State
	YES Connectable Advertising State
	YES Directed Advertising State
	YES Passive Scanning State
	YES Active Scanning State
	YES Initiating State/Connection State in Master Role
	YES Connection State in the Slave Role
	YES Non-connectable Advertising State and Passive Scanning State combination
	YES Scannable Advertising State and Passive Scanning State combination
	YES Connectable Advertising State and Passive Scanning State combination
	YES Directed Advertising State and Passive Scanning State combination
	YES Non-connectable Advertising State and Active Scanning State combination
	YES Scannable Advertising State and Active Scanning State combination
	YES Connectable Advertising State and Active Scanning State combination
	YES Directed Advertising State and Active Scanning State combination
	YES Non-connectable Advertising State and Initiating State combination
	YES Scannable Advertising State and Initiating State combination
	YES Non-connectable Advertising State and Master Role combination
	YES Scannable Advertising State and Master Role combination
	YES Non-connectable Advertising State and Slave Role combination
	YES Scannable Advertising State and Slave Role combination
	YES Passive Scanning State and Initiating State combination
	YES Active Scanning State and Initiating State combination
	YES Passive Scanning State and Master Role combination
	YES Active Scanning State and Master Role combination
	YES Passive Scanning State and Slave Role combination
	YES Active Scanning State and Slave Role combination
	YES Initiating State and Master Role combination/Master Role and Master Role combination
```

Unfortunately when doing a scan I cannot find any device
```
sudo hcitool lescan
LE Scan ...
```
and trying to connect to my WeDo 2.0 SmartHub fails:
```
sudo gatttool -b A0:E6:F8:1E:58:57 -I
[A0:E6:F8:1E:58:57][LE]> connect
Attempting to connect to A0:E6:F8:1E:58:57
Error: connect: Operation not supported (95)
```

It is very similar to this (but not sure if it is the same Texas chipset):
https://e2e.ti.com/support/wireless_connectivity/wilink_wifi_bluetooth/f/307/t/217059

Are you guys aware of any firmware for the BT chipset or any hci command that might enable LE communication?

dlech, 2016-07-31:
According to the [TI website](
http://www.ti.com/product/cc2560) this chip does not support BLE.

The chip in the EV3 is the CC2560. The version with BLE support is CC2564.

dlech, 2016-07-31:
If you want built-in BLE, you will have to replace the entire chip :wink:

JorgePe, 2016-07-31:
I was afraid of that.
But why does the bluez stack recognizes LE features after running those commands?  I just hammered some kind of flags?

dlech, 2016-07-31:
Sounds like a question for the linux bluetooth mailing list.

JorgePe, 2016-08-02:
OK, I close this one, sorry to waste time.
But I still don't understand LEGO option - if the CC2564 is pin compatible and with WeDo 2.0 / Power Functions 2 / Mindstorms "next" being prepared since at least two years it would make sense to include native BLE support.

closed 2016-08-02 by JorgePe