ev3dev/ev3dev #712: Setting a bluetooth connection between Arduino uno and the EV3

volansky, 2016-08-01:
Hey guys, I wanted to consult with you before moving on with my project* (hope that's alright with you).

I was able to connect the ev3dev (nightly) with win (10) and VM (trusty), compile a C program and run it. now I want to establish a bluetooth communication and test it between the EV and an arduino Uno (using HC-05/HC-04).
The way I see it I can either configure the connection through brickman and terminal commands or I can connect them using a C program (which is going to run anyway).

My question is which way is more suitable for my needs (creating a plug&play experience for the end user and for a two way communication with an Arduino)?

* Link to the project explnation - https://github.com/ev3dev/ev3dev/issues/703#issuecomment-234782536

dlech, 2016-08-01:
I think the way you are doing it now makes sense. On a smart phone, you always pair the device using the generic Bluetooth settings first. Then the application using the bluetooth device takes care of the rest. Likewise, I think it makes sense to pair your device using Brickman and then let your application do the rest.

volansky, 2016-08-04:
Just to be sure, I can run a bash to automate the connection on start right?

Also, I'm trying to work with the bluetooth but it dosen't go very well..
When connecting to devices via brickman I get this message : GDBus.Error:org.bluez.Error.NotAvailable: Operation currently not available.
And when trying to use bluetooth commands on the terminal it dosen't reconize them.

dlech, 2016-08-04:
Tip: use the `bluetoothctl` command. It is easier.

What type of service are you trying to connect? The connect button in brickman only connects a few well known services that are built in to bluez (e.g. HID for keyboard/mouse/gamepad). If the bluetooth service you are using is not one of these, then you will have to write a custom application to handle the connection.

volansky, 2016-08-04:
I'm not quite sure what service im trying to connect as I just now learned that there are differences, as I explained before I want to create a physical blocks that will send a block of commands for the brick to follow and responed when ever a command is executed, I guess I'll need to write my own service for that..
But right now I'd sattle with sending a char to the arduino from the brick just to see an esablished connection, I got them paired but when trying to connect with the bluetoothctl connect command I get:

> [bluetooth]# connect 20:16:03:21:82:87
> Attempting to connect to 20:16:03:21:82:87
> [CHG] Device 20:16:03:21:82:87 Connected: yes
> Failed to connect: org.bluez.Error.NotAvailable
> [CHG] Device 20:16:03:21:82:87 Connected: no

Any idea?

P.S thanks for the tip!

dlech, 2016-08-04:
Run `info 20:16:03:21:82:87` in `bluetoothctl` to see what services are listed.

volansky, 2016-08-04:
[bluetooth]# info 20:16:03:21:82:87
Device 20:16:03:21:82:87
        Name: HC-05
        Alias: HC-05
        Class: 0x001f00
        Paired: yes
        Trusted: yes
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)

dlech, 2016-08-04:
So you need to write a program that connects to the serial port service (commonly called RFCOMM). There are some command line tools to do this too. It has been so long since I used them, I don't remember what they are called or how to use them.

dlech, 2016-08-04:
Try `rfcomm --help`

dlech, 2017-08-12:
Is this still and issue? Can this be closed?

volansky, 2017-08-13:
Yea it can be closed, thanks.

closed 2017-08-13 by dlech