ev3dev/ev3dev #713: What is the recommended route to developing a new language interface.  I see many varying paths in my ev3 browsing.

tbrowder, 2016-08-01:


rhempel, 2016-08-01:
The preferred route is through our `autogen` system here:
(http://ev3dev-lang.readthedocs.io/en/latest/autogen.html)

It's a bit of a grind to get started, but it makes for a lot less cut/paste when you're developing.

Usually I start with getting a simple function to read/write a device attribute, and then turn that into a `liquid` template. If the language supports objects, you'll want a generic device interface too.

A good starting point is (https://github.com/rhempel/ev3dev-lang-python)

dlech, 2016-08-01:
I think the best place to start is understanding how the ev3dev drivers work. For example there is a [motor tutorial](http://www.ev3dev.org/docs/tutorials/tacho-motors/) that walks you through understanding how the tacho-motor subsystem works.

Once you understand how things work from the command line, you can translate them to whatever language you are using.

Also, I think you mentioned somewhere that you were interested in perl. @JanderII has [done some work with ev3dev and perl](https://www.facebook.com/groups/legomindstorms/permalink/570013999813497/). Maybe you could collaborate.

closed 2019-03-25 by dlech