ev3dev/ev3dev #714: Is there a language agnostic test suit for a language interface?

tbrowder, 2016-08-01:


rhempel, 2016-08-01:
Not really - and I'm not sure what that would look like if it existed. I have a (tentative) start on a test interface for Python [here](https://github.com/rhempel/ev3dev-lang-python/tree/develop/tests/motor)[here]  but I'm not really happy with it.

As I mentioned in #713 - if you use `autogen` then almost all of the code is generated for you once you have the liquid templates working. From that point on, if you can read/write the motor `position` attribute it's a pretty good bet you can read a sensor attribute.

rhempel, 2016-08-01:
The Python test suite I have is intended to test the drivers, not the Python binding - although it does test the binding as a side effect.

tbrowder, 2016-08-01:
On Monday, August 1, 2016, Ralph Hempel <notifications@github.com> wrote:

> The Python test suite I have is intended to test the drivers, not the
> Python binding - although it does test the binding as a side effect.
>
Thanks, Ralph!

>

dlech, 2016-08-01:
> and I'm not sure what that would look like if it existed.

It would like like our user defined sensor driver. You can programatically define any sensor and its behavior this way. Of course this is not ready to use at this time, but that is what it would look like.

dlech, 2017-10-28:
There is https://github.com/ev3dev/ev3dev-lang-fake-sys

It doesn't really do much though.

closed 2019-03-25 by dlech