ev3dev/ev3dev #715: EV3 medium motor does not support all the commands except run-direct

WayneLow, 2016-08-03:
### System information:

* I am using: *LEGO MINDSTORMS EV3*

* I installed ev3dev using this image file: *ev3-ev3dev-jessie-2015-12-30.img*

* My kernel version is: *4.4.13-12-ev3dev-ev3+*

* My host computer is running: *ubuntu 16.04*

* I am connecting using: *USB*

### About my issue:

EV3 Medium Motor is not able to use commands other than run-direct.
It does not support run-forever, run-to-abs-pos, run-to-rel-pos and run-timed

I tried command before in Large Motor it is working but dint manage to work in Medium Motor.
```
echo '3000' > time_sp
echo '75' > duty_cycle_sp
echo 'run-timed' > command
```

WasabiFan, 2016-08-03:
The reason that the `run-direct` command is the only one that is successfully running the motor is that it is the only one that gets the target speed/power from the `duty_cycle_sp` attribute. For the other commands, the attribute that you want is `speed_sp`. Unlike the duty cycle, the speed value is in the range from 0 to the value of `max_speed`, or a negative value in the equivalent range to go in reverse. It is _not_ a percentage.

If you haven't seen it already, the [motor tutorial](http://www.ev3dev.org/docs/tutorials/tacho-motors/) should explain motor usage a bit better.

From the [motor docs page](http://www.ev3dev.org/docs/drivers/tacho-motor-class/) (when referring to the `speed_sp` attribute):

> Writing sets the target speed in tacho counts per second used for all `run-*` commands except `run-direct`. Reading returns the current value. A negative value causes the motor to rotate in reverse with the exception of `run-to-*-pos` commands where the sign is ignored. Use the `count_per_rot` attribute to convert RPM or deg/sec to tacho counts per second.

You said that the commands given worked for a large motor but not a medium one; I would not expect it to work for either of them, so if it did work something odd is probably going on.

Some background on duty cycle and speed: Duty cycle is, in essence, a measure of the amount of power being sent to the motor. The duty cycle does not directly control the physical speed of the motor; if the duty cycle were 50% and you tried to stop the motor with your hand or an assembly attached to it, it would just slow down. Meanwhile, speed is a value indicating the physical speed of the motor, and the drivers will do everything they can to keep the motor running at the speed that you select. Due to some complexities in handling ramping and positioning, only speed values are supported for most commands.

---

@dlech We need to update the docs so that this information is clearer, especially in the descriptions for the commands. It would also be great if the motor class documentation were more easily discoverable.

WayneLow, 2016-08-03:
@WasabiFan Thanks for the swift reply. Your clarification on the difference between Duty Cycle and Speed is good. Perhap you can add the limitation of duty_cycle in the documentation.

closed 2016-08-03 by WayneLow

WasabiFan, 2016-08-03:
Great! I'm happy to clear things up. I will make sure to update the docs when I can to make this clearer.

dlech, 2016-08-03:
> We need to update the docs so that this information is clearer

Open a new issue so we don't forget.