ev3dev/ev3dev #716: Add better documentation on speed/duty cycle

WasabiFan, 2016-08-03:
We need to update the docs so that the distinction between duty cycle and speed is clearer, especially in the descriptions for the commands. It would also be great if the motor class documentation were more easily discoverable.

BenTristem, 2016-08-07:
I'm new to this, and I agree. I'm using ev3-ev3dev-jessie-2015-12-30 and I can't seem to make the motor's run using my RunTimed method (RunSleep works fine)...

`def RunSleep(motorName, speed, seconds):
    m = Motor(motorName)
    m.run_forever(duty_cycle_sp = speed)
    time.sleep(seconds)
    m.stop()

def RunTimed(motorName, speed, seconds):
    m = Motor(motorName)
    m.run_timed(duty_cycle_sp = speed, time_sp = seconds * 1000)`

Any help appreciated

dlech, 2017-10-28:
Any specific suggestions for http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/motors.html#tacho-motor-subsystem?