ev3dev/ev3dev #718: USB and PS3 gamepad support in EV3Dev?

cellchow, 2016-08-05:
I am trying to make use an USB gamepad to control my robot. But the Python code that I write cannot detect it. I am connecting it to the EV3 via a USB hub.

The following is my code, a very simple one

```
import evdev

def deviceListing() :
	print("That is all the deveices that I can detect")
	devices = [evdev.InputDevice(fn) for fn in evdev.list_devices()]
	for device in devices:
		print(device.name)

deviceListing()

```
The result looks like that 

That is all the devices that I can detect
EV3 speaker beep
EV3 buttons

But when I lsusb it, the gamepad will show up

Bus 001 Device 004: ID 046d:c21d Logitech, Inc. F310 Gamepad [XInput Mode]
Bus 001 Device 003: ID 0bda:8178 Realtek Semiconductor Corp. RTL8192CU 802.11n WLAN Adapter
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. 4-Port HUB
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

If I dmesg it, I can see the gamepad too

[   45.433180] usb 1-1.3: Product: Gamepad F310
[   45.433216] usb 1-1.3: Manufacturer: Logitech
[   45.433250] usb 1-1.3: SerialNumber: E14CB527
[   47.632418] usbcore: registered new interface driver rtl8192cu
[   54.998437] cfg80211: Calling CRDA to update world regulatory domain

But if I do these, the gamepad and the js0 are not there

robot@ev3dev:~$ ls /dev/input
by-path  event0  event1
robot@ev3dev:~$ ls /dev/input/event*
/dev/input/event0  /dev/input/event1

And if I check the loaded module, the module joydev.ko is not loaded

Any ideas I can fix the problem? I also tried to use my PS3 gamepad, and it does not work either. I make use of the USB connection first, it won't work; and I make use of the bluetooth connection, it won't connect. I followed the instructions that is given by 
[(http://www.ev3dev.org/docs/tutorials/using-ps3-sixaxis/)], 
but it just does not work.

The EV3Dev that I am using is 

Linux ev3dev 3.16.7-ckt21-9-ev3dev-ev3 #1 PREEMPT Tue Dec 15 15:16:17 CST 2015 armv5tejl GNU/Linux

Thank you very much

Best

JorgePe, 2016-08-05:
Hi!

It's been a while since I last used my gamepad with the ev3dev. I have [this old post](http://ofalcao.pt/blog/2014/ev3dev-wireless-gamepad)) , unfortunately it's in portuguese, I hope you can understand the shell commands and the scripts.

When you plug the USB gamepad do you see any OHCI messages in dmesg, like

```new full-speed USB device number 4 using ohci```

If yes, you can try

```root@ev3dev:~# apt-get install joystick
root@ev3dev:~# jstest --normal /dev/input/by-id/....```

At that time I had problems with jstest compatibility with ev3dev kernel but at least got some output, don't know if problems are fixed in present kernel. Then I used pygame, there is a script at the end of the post that gives information about the gamepad, if it works you can identify all buttons codes with it.

dlech, 2016-08-05:
`joydev` has long been deprecated, so has not enabled in the ev3dev kernel because we haven't needed it.

It looks like you can try switching the controller to "D" mode or we need to enable the `xpad` driver in the kernel.

dlech, 2016-09-04:
Fix released in `14-ev3dev` kernel.

closed 2016-09-04 by dlech

cellchow, 2016-09-07:
Thanks for the help.
But may I ask ......

1. What is the direct link to the new kernel?
2. After I download it, how do I merge it to my existing system? Can you send me some instructions to get me started?
3. With the new kernel work with TP-Link wifi adapter?

dlech, 2016-09-07:
It would probably be best to download the latest image from https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/ and re-flash your SD card.

Otherwise if you have several hours to spare, you can upgrade everything. Make sure your EV3 is connected to the Internet, then run...

    sudo apt-get update
    sudo apt-get upgrade
    sudo apt-get dist-upgrade
    sudo reboot

If your wifi adapter is working now, it should still work with the new kernel. If it is not working, then it still won't work.

cellchow, 2016-09-07:
Worked right out of the box! Thanks for the help!