ev3dev/ev3dev #719: Hi Technic IRLink question

JorgePe, 2016-08-08:
Hello.
I'm using EV3 with nightly build 2016-07-26 and a few  updates after that (but not as of today, I have no Internet access for my EV3 today), kernel is "ev3dev 4.4.15-13-ev3dev-ev3"

I have a HT IRLink, it is connected to Input Port 1, dmesg:

```
lego-port port0: Added new device 'in1:nxt-i2c-host'
i2c-legoev3 i2c-legoev3.3: registered on input port 1
```

Not sure if this is the proper way to initialize:
```
sudo sh -c "echo ht-nxt-ir-link 0x01 > /sys/bus/i2c/devices/i2c-3/new_device"
```

dmesg:
```
lego-sensor sensor0: Registered 'ht-nxt-ir-link' on 'i2c-legoev33:i2c1'.
i2c i2c-3: new_device: Instantiated device ht-nxt-ir-link at 0x01
```

And now I do have a sensor0:
```
cat /sys/class/lego-sensor/sensor0/mode
IRLINK
```

The problem is that I can only read value0, not write to it. How can I send commands to a Power Functions IR receiver?

dlech, 2016-08-08:
> Not sure if this is the proper way to initialize

It should be automatically detected. Can you run this (without configuring `new_device` first)?

    i2cdump -y 3 0x01

and post the output here

> How can I send commands to a Power Functions IR receiver?

Use the [direct](http://www.ev3dev.org/docs/sensors/using-i2c-sensors/#direct-reading-and-writing-of-the-sensor) attribute.

Another option is to not use the `lego-sensor` class and instead write I2C messages directly. I need to update the [docs](http://www.ev3dev.org/docs/sensors/using-i2c-sensors/#direct-i2c-communication-going-driverless), but there is a new `other-i2c` mode for port that you can use. It sets up the I2C adapter but does not attempt to probe the hardware.

JorgePe, 2016-08-08:
```
robot@ev3dev:~$ sudo i2cdump -y 3 0x01

No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
10: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
20: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
30: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
40: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
50: XX XX XX ff XX XX XX XX XX XX XX XX XX XX XX XX    XXX.XXXXXXXXXXXX
60: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
70: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
80: XX XX XX XX XX XX 23 XX XX XX XX XX XX XX XX XX    XXXXXX#XXXXXXXXX
90: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
a0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
b0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
c0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
d0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
e0: XX XX XX XX XX XX XX XX XX XX XX XX XX 00 00 XX    XXXXXXXXXXXXX..X
f0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
```

dlech, 2016-08-08:
It looks like you sensor is busy. There should be numbers where the `XX` are. Unplug the sensor and plug it back in to make sure that the `lego-sensor` class driver is not loaded, then try again.

JorgePe, 2016-08-08:
I considered other-i2c but [this page](http://www.ev3dev.org/docs/ports/legoev3-input-port/) says "Configure for I2C communications but do probe for NXT sensors". So it's a typo and should say "do not probe" ?

dlech, 2016-08-08:
> So it's a typo and should say "do not probe" ?

Yes, that looks like a typo.

JorgePe, 2016-08-08:
After unplug/replug gives same "XX" output.

If I boot without the IRLink and then connect (IN 1 again) I also get the same.

dlech, 2016-08-08:
Perhaps that is why it is not being automatically detected. Any luck with the `other-i2c` mode?

JorgePe, 2016-08-08:
If you mean
```
echo other-i2c > /sys/class/lego-port/port0/mode
```
the i2cdump output is similar.
As for using direct I need to read a bit more to understand what to send to the IRLink.

JorgePe, 2016-08-09:
If I boot from a non-nightly build SD card it is not autodetected but i2cdump works:
```
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: fd 56 31 2e 32 20 20 20 48 69 54 65 63 68 6e 63    ?V1.2   HiTechnc
10: 49 52 4c 69 6e 6b 20 20 00 00 00 00 00 00 00 00    IRLink  ........
20: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
80: fd 56 31 2e 32 20 20 20 48 69 54 65 63 68 6e 63    ?V1.2   HiTechnc
90: 49 52 4c 69 6e 6b 20 20 00 00 00 00 00 00 00 00    IRLink  ........
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
```

```
uname -a
Linux ev3dev 4.4.13-12-ev3dev-ev3 #1 PREEMPT Tue Jun 14 20:26:05 CDT 2016 armv5tejl GNU/Linux
```
I don't remember the last time I updated this SD card but I'm quite sure it was updated in July when I was testing my WeDo2 scripts.

JorgePe, 2016-08-09:
I did "update & dist-upgrade" and after reboot i2cdump output is now all "XX" again:

Only these packages were changed:

- base-files
- linux-image-4.4.15-13-ev3dev-ev3
- rtl8188eu-modules-4.4.15-13-ev3dev-ev3
- linux-image-ev3dev-ev3

JorgePe, 2016-08-09:
If I boot from "ev3-ev3dev-jessie-2015-12-30.img.xz" (no updates at all) I see in dmesg that auto detection worked and I have "/sys/class/lego-sensor/sensor0" right after login.

dlech, 2016-08-09:
OK, to recap:

* auto detection works on `9-ev3dev` kernel
* auto detection does not work, but i2cdump works on `12-ev3dev` kernel.
* neither auto detection or i2cdump work on `13-ev3dev` kernel.

correct?

dlech, 2016-08-09:
Also, do you have problems with other I2C sensors or just the IRLink?

JorgePe, 2016-08-09:
That is correct.
I also have auto-detection problems with latest nightly build and NXT Temperature sensor, need to
```
sudo sh -c "echo lego-nxt-temp 0x4C > /sys/bus/i2c/devices/i2c-4/new_device"
```
Before initializing I tried "i2cdump -y 4 0x04C" with strange results but I know it is working because  I can read the temperature (I'm using it to turn on my Dyson fan through a BT4 power plug).

dlech, 2016-08-09:
Please don't say "latest nightly build" without also specifying something more exact. It will only be latest for a short time and will lead to confusion. Kernel versions are much more useful.

I wonder if there was some upstream kernel change that is contributing to this.

dlech, 2016-08-09:
I also wonder if it could be #77 is not fixed after all. You might be able to tell if there are sound playback issues as well.

You can also try adding the `console=ttyS1,115200` kernel option to `/etc/default/flash-kernel` then run `sudo flash-kernel` and reboot (do this on a nightly image).

JorgePe, 2016-08-09:
Sorry, you're right... it should be 2016-07-26 build but with all these SD cards changes I boot again to check:

Auto-detection of nxt-tmp doesn't work with these two kernels:

```
Linux ev3dev 4.4.13-12-ev3dev-ev3 #1 PREEMPT Tue Jun 14 20:26:05 CDT 2016 armv5tejl GNU/Linux
Linux ev3dev 4.4.15-13-ev3dev-ev3 #1 PREEMPT Sun Jul 24 17:54:14 CDT 2016 armv5tejl GNU/Linux
```

I added the console option to the "ev3dev 4.4.15-13-ev3dev-ev3" image.

After reboot, both IRLink and NXT Temp sensors were auto-detected. Don't understand your voodoo but it works.

dlech, 2016-08-09:
I don't understand it either. It makes no sense that enabling console on a serial port would effect I2C. The two systems are completely unrelated, both in hardware and in software (as far as I know).

An unfortunate side effect of `console=ttyS1,115200` is that it interferes with UART sensors and Bluetooth, which is why we removed it. (#623, #572).

I will be working on an alternate implementation for I2C in the near future (ev3dev/lms2012-compat#19). We should be able to port this to the EV3, so maybe we will get it sorted (at least the I2C part) fairly soon.

JorgePe, 2016-08-10:
Thanks.
About the IRLink, I figured how to use direct attribute and also found some Java and C (RobotC) examples, I find it difficult to understand but will try.

Yesterday I also found Lua examples from @rhempel but today his site is down.

JorgePe, 2016-08-11:
I figured how to send a Power Functions command through the IRLink using python.
Still have to make a python script to accept arguments instead of just sendi fixed / predefined commands but at least I can finally see a motor running.

```
#!/usr/bin/python
import sys
from time import sleep

# ComboPWM mode, motor A = Motor B = Full Forward
pfCommand="\x80\x90\x49\x20\x82\x09\x04\x10\x41\x20\x82\x11\x02\x01"

register = 0x42

while True:
  for i in range(5):
    fd = open("/sys/class/lego-sensor/sensor0/direct", 'wb',0)
    fd.seek(register)
    fd.write(pfCommand)
    fd.close()
    sleep(0.004)
  sleep(0.060)
```
(NOTE: timings DO NOT full respect Power Functions specs; NO initialization is done on the IRLink; NO check is made at I2C level)

Thanks for the tip about the "direct" attribute.

dlech, 2016-09-04:
Closing since this is essentially a duplicate of #47. Should be fixed without the workaround needed in the next kernel release.

closed 2016-09-04 by dlech