ev3dev/ev3dev #722: .net support for .net languages?

tim241, 2016-08-14:
Is there any .Net language released yet? I really want mono for the lego EV3 with up to date support, Monobrick is outdated and has problems. So is there any mono for the ev3 in ev3dev?

dlech, 2016-08-14:
There was some effort a while back, but it seems stalled: https://github.com/ev3dev/ev3dev-lang/pull/129

Personally, I am waiting for https://github.com/dotnet/corert to get far enough along to be usable on the EV3. The corert does not have direct support for ARMv5 (which is what the EV3 CPU is), but only ARMv7 (e.g. Raspberry Pi 2/3). However, there is the possibility of generating C++ code from the .NET compiler. Then the generated C++ code can be compiled for the EV3.

tim241, 2016-08-14:
Ok, I really hope that I can program the lego Ev3 in vb.net :)

tim241, 2016-08-18:
So how can I install that C# for ev3dev? it is an pull request :+1:

tim241, 2016-08-18:
This is intresting: http://bleedingnedge.com/2015/11/08/asp-net-5-on-lego-mindstorms-ev3-using-ev3dev/

WasabiFan, 2016-08-18:
That blog post is the one written by the same guy that opened the PR linked to above; those instructions should still work. Keep in mind, though, that the PR code will not work on recent ev3dev releases because it hasn't been updated to keep up with driver changes.

tim241, 2016-08-18:
Sorry, I am not a linux expert , I tried making android kernels but I am terrible at it. Otherwhise I should create it by myself :(

tim241, 2016-08-18:
Then I will wait

tim241, 2016-08-18:
I will probably just try the C# with ev3dev it when I am home tomorrow.

closed 2016-09-13 by tim241