ev3dev/ev3dev #726: ev3dev sound - support for MIDI in ALSA?

JorgePe, 2016-08-21:
Hello.

I think I might be pushing the limits of the EV3 but I'm trying a laser harp. I can't play anything but my wife has some music theory background and plays piano, I gave her a MIDI keyboard some years ago and she used it to generate scores in an UbuntuStudio computer, it was her first contact with Linux and my first contact with the linux sound system.

ev3dev kernel: Linux ev3dev 4.4.17-14-ev3dev-ev3

I already have 6 EV3 color sensors (and a MindSensors MUX, another one in the way for a 7th sensor and an ultrasonic sensor to read the distance of the hand).

I'm using python.

I can already read the 6 sensors in python. I can play tones (with Sound.tone) but of course it doesn't sound like an harp. So I tried Sound.play with some harp samples I found on the net. Sounds much better but samples are to long so the "artist" have to wait for the end of each note before moving the hand to another "string". And it doesn't allow multiple notes at the same time.

So I went for MIDI. It has polyphony, there are some python libraries and since it uses soundfonts
I can easily change the instrument (me and my wife had the idea to use NFC tags to identify the "artist" or the "instrument").

 I installed timidity++ (a software MIDI synth) but id doesn't work, it seems to need a module snd_seq that doesn't exists.

Then I went for fluidsynth and mingus, a python library that works with fluidsynth and ALSA. The soundfount that cames with fluidsynth is to big to load on memory but I found two small soundfonts that load fine. In Ubuntu I can play some nice notes, even several at same time. But in ev3dev, when mingus initialize fluidsynth, I immediately get a clicking noise (even before playing a note)

```
from mingus.midi import fluidsynth
from mingus.containers.note import Note
fluidsynth.init("Concert_Harp.sf2", "alsa")
# instrument 46 is Harp
fluidsynth.set_instrument(1, 46)
fluidsynth.play_Note(Note("C-3"))
```

I suspect fluidsynth also needs ALSA snd_seq module but found no evidences.

So...
- anyone ever tried midi with ev3dev? Or some other method for polyphonic sound?
- is it possible to include ALSA snd_seq module?

My "Plan B" is using the BrickPi since it supports the EV3 color sensor (but not sure if also supports the MindSensors MUX).

"Plan C" is much more drastic - use the EV3 as a MIDI instrument and send the MIDI commands to a Raspberry Pi running timidity++ or fluidsynth.

JorgePe, 2016-08-21:
Update: fluidsynth seems to need snd_seq_hw

If I start just fluidsynth it also makes click noises but I have this output:
```
robot@ev3dev:~$ fluidsynth --audio-driver=alsa Concert_Harp.sf2
(...)
ALSA lib seq_hw.c:457:(snd_seq_hw_open) open /dev/snd/seq failed: No such file or directory
fluidsynth: error: Error opening ALSA sequencer
Failed to create the MIDI thread; no MIDI input
will be available. You can access the synthesizer 
through the console.
```

dlech, 2016-08-21:
I'll turn on so MIDI stuff in the next kernel release and we will see what happens. I've always wanted to play around with MIDI on the EV3, but just never got around to it, so I'm curious to see what you come up with.

JorgePe, 2016-08-21:
That's great, thanks!!

I also discovered that ALSA has "aseqnet" and "aconnect" commands that allows a client to send MIDI to an host but it also requires snd_seq so every alternative I find ends on ALSA MIDI support.

If it doesn't work I can always use MQTT or some other network protocol to send the sensor values to my laptop or to a Raspberry Pi and play the music there. Even with some latency it will probably sound good enough (considering it's LEGO, of course).

I saw somewhere in ev3dev' github that MIDI USB support is/was included. So I will also try my wife's MIDI Keyboard...  if it works I might have to buy an EV3 just for my wife :)

JorgePe, 2016-08-23:
A quick demo using MQTT to send the notes to my Ubuntu laptop:
https://youtu.be/eKW4_QbIFak

I'm afraid the EV3 might not have enough power to run a MIDI soft synth but I hope to at least use ALSA to send the MIDI commands and drop the MQTT layer.

dlech, 2016-09-04:
Fix released in `15-ev3dev` kernel.

closed 2016-09-04 by dlech

JorgePe, 2016-09-05:
@dlech I confirm, after tonight update to '4.4.19-15-ev3dev-ev3' I have ALSA MIDI and I can run timidity and fluidsynth without errors related to missing modules.
No sound whatever but that's another question :)

dlech, 2016-09-05:
timidity file.mid --output-mono --sampling-freq=22500

\o/

JorgePe, 2016-09-05:
It works!!!
"Prepare for awesomeness"

JorgePe, 2016-09-05:
adding this entry to '/etc/timidity/timidity.cfg'
`opt s22.5kHz`
is enough. There are other entries there for reducing CPU usage and choose soundfont, will try later.

JorgePe, 2016-09-05:
Sorry to keep writing to a closed issued... just a last update (and thanks for advertising my harp at the front page)

fluydsynth seems to heavy for the EV3, I can start it from command line and load a small soundfont but there's to much clicking and when starting it from python with mingus I cannot pass parameters to reduce sampling rate, it doesn't work at all.

I installed python-rtmidi and mido and managed to choose the instrument and play notes through timidity in a similar way to mingus through fluidsynth. Sound is acceptable but latency is awful - each note plays almost 15 seconds after the instruction. I uncommented all performance options in timidity.cfg and even started timidity manually so I can tell it to use mono but no luck.

So it seems that programatically using MIDI on the EV3 is really to much and my approach with MQTT is still the best, at least for now - mido also has a MIDI client/server feature, will try it one of these days.

Nevertheless having MIDI support in ev3dev is still a great thing, thanks for the effort!

dlech, 2016-09-05:
The 15 second latency must be from a new python process starting each time or something like that. I connected my keyboard via USB MIDI and used it to play sounds. The latency was still annoying, but less that 1 second. There is also a bunch of clicking due to a defect in the ev3dev sound driver. I've fixed this on BeagleBone/EVB but haven't backported it to the EV3 yet.

It will probably work better the other way around though. If you have a MIDI device like a keyboard, you can have the EV3 send MIDI messages to the other device and have the sound generated there. If you don't have any MIDI devices, a BeagleBone would make a nice substitute. BeagleBone has a realtime kernel for low latency and `modprobe g_midi` should turn the BeeagleBone into a USB MIDI device via the peripheral port.

JorgePe, 2016-09-05:
I have a MIDI (USB) keyboard. It's just input, no output. But if the kernel recognizes it, I'll try to find it a usage as a controller for some animatronics.

Perhaps the 15 seconds are from the way mido uses python-rtmidi. I'll try just python-rtmidi.

JorgePe, 2016-12-24:
Sorry to revive a closed issue but if someone ever gets here looking for MIDI support... ev3dev works now quite well as a network MIDI input/output device with ALSA aseqnet.
I've made a MIDI robot that uses ALSA aseqnet to receive MIDI commands from my laptop and a pygame-based script to parse those commands, reacting to 3 velocity (intensity) on 3 different channels.
It's not S3NSORH3ADS robotic band but I'm pretty happy with my [ANG3L](https://youtu.be/VlvRxHdqeto)

Thank you guys and Merry Xmas!

Tooshe, 2019-07-23:
Hello, I'm new to ev3dev and python and have encountered a problem with mido.

import mido 
ImportError: No module named 'mido'

Mido is installed correctly and I can enter it in the terminal >>> without the error but Visual Studio Code always gives me the error.

Is it possible to use mido with ev3dev? 
Am I doing something wrong?

I'd really like to spin some motors from midi file data but I've hit a brick wall, could someone point me in the right direction?

benwiley4000, 2021-01-30:
@JorgePe Hi I'm having some issues running fluidsynth. I noticed a couple warnings when I start up:

```
fluidsynth: warning: Requested a period size of 64, got 4095 instead
fluidsynth: warning: Failed to set thread to high priority
```

That's just with `fluidsynth -a alsa`.

I'm also running out of memory when I try to load a soundfont. I do hear a small click when the program starts but I wasn't successful in sending any `noteon` / `noteoff` events (it's possible I'm misunderstanding how that's supposed to work). For now I'm just using bash, no python-specific SDKs or anything, and not trying to use any midi controller or the like - just a terminal.

Any ideas what I might be doing wrong? Did you run fluidsynth with any special args to allow it to run correctly on the ev3? Thanks in advance.

JorgePe, 2021-01-30:
Very very sorrow but I cannot find my notes from that time.

I  do remember trying fluidsynth but I don't remember ever made it work. Timidity worked (as above).
Also fluidsynth is too demanding - even timidity with a big soundfont is. If you have a USB sound card, you can use less CPU using it instead of ev3dev ALSA sound card - you also get stereo sound, 16 bit@44 kHz sound much better than EV3 speaker and you can even use an audio amplifier. Another option could be BT audio, it is possible to pair ev3dev to a BT speaker but I don't do this anymore so not sure how to do it with current ev3dev version.

But I returned to MIDI recently, thanks to David inclusion of MIDI on ev3dev kernel so there are other options you can explore to get MIDI sound. If you explain me what you pretend, we can try to find a solution.

For pure MIDI sound locally on the EV3 best option is a USB-MIDI adapter and a MIDI synth. I played good MIDI musics this way using just 4~5% of CPU (no soundfonts required, no softsynth required, not even audio card required).

If you don't want to spend money on this option, you can send MIDI messages to your own "MIDI gateway". My first version of the Laser Harp was too clumsy but I used my linux laptop with fluidsynth and used MQTT to transfer messages from the EV3's to the laptop.

Now I am using pure MIDI over Wi-Fi (ipMIDI or mutlicast MIDI). I have a Raspberry Pi receiving MIDI messages connected to a MIDI synth but it can also use a soft synth - look for zynthian or pacthbox OS projects for inspiration. Latency can be a problem, I am using a dedicated Wi-Fi router for this... it works great: https://youtu.be/dTuFZ_FHsHQ

In fact it works so great that my family is planning a LEGO jam session in a few days with some LEGO MIDI instruments and a MIDI keyboard. Only problem is my total lack of music skills :D

benwiley4000, 2021-01-30:
So basically I would only need to load one voice/instrument at a time and I
would only play one note at a time. Each time you start a new note, the
last one is cancelled first. Monophonic synthesizer basically. I'm willing
to use any tool for the job. A basic midi soundbank is fine, doesn't
necessarily need to be SF2. I would just like something more expressive
than `beep`. The most important requirement that makes it impossible to use
beep or similar is that I don't know how long the note will last until the
note is lifted by the user, so I need to probably use the midi event system.

Does that info help?

Thanks

JorgePe, 2021-01-30:
Yes it helps, thank you.
Yes, if the note cannot be interrupted (like an organ or a flute, I think) MIDI is the easier option (other option would be to generate a PWM signal to replace the ev3dev python sound module but only useful if you know you are using just a few notes).
Timidty in alsa mode is you best option, at least for a start until you get something solid.
To generate MIDI messages to send to timidity you can use a python library but most of them are heavy and take a lot of time to install (if you can install because of some dependencies might not be available for armel). I got a simpler (?) way that I am using with pybricks (so micropython) but can also be used with ev3dev2 python: I install 'amidicat' that creates a virtual MIDI device and allows using a file pipe as input - so you start 'amidicat' and use 'aconnect' to route it to 'timidity', then on your python script you open a file (the pipe) and write midi hexadecimal messages to it (for instance, if you write "99 3C 64"  it means NOTE ON note=C4 velocity=64)

See here for more details, feel free to reach me there for more details and please keep us informed of any progress, not used to see people using MIDI with the EV3
https://github.com/JorgePe/randomideas/tree/master/MIDI/DrumKit

Be prepared to send a "all notes off" or "panic" message. Sometimes MIDI doesn't work correctly and if a NOTE OFF is missed it will play forever. Even commercial MIDI controllers or App use to have a panic button because of this "feature".

Final note: I now zero of music. Been learning MIDI through Google so I am not an expert, my wife gives some practical advices, she now how to read scores and plays piano. So if you do know someone with musical skills, ask him/her for help to :D

benwiley4000, 2021-01-30:
> Final note: I know zero of music. Been learning MIDI through Google so I

I'm a musician/synth player so hopefully I'll be able to manage. :)

Thanks for all the tips! I'll try piping input to timidity. What args did
you pass to timidity?

JorgePe, 2021-01-30:
I think just ' timidity -Os -iA &' like I do in Ubuntu or perhaps just 'timidity -iA'.
There is also conf file, I commented almost everything to reduce overhead (thinks like reverb).
There are some free General MIDI soundfonts available, not sure if they came with timidity (in fact timidity++) or you need to install with "sudo apt install fluid-soundfont-gm" and then edit the config file.
When it is working, you have 4 ports:
```
TiMidity starting in ALSA server mode
Opening sequencer port: 128:0 128:1 128:2 128:3
```
and you can test if with 'aplaymidi -p 128:0 somemidifile.mid'

benwiley4000, 2021-01-30:
Amazing thanks for all your help. I'll try it out this weekend.

dlech, 2021-01-30:
> What args did you pass to timidity?

https://github.com/ev3dev/ev3dev/issues/726#issuecomment-244650161

benwiley4000, 2021-01-30:
Thanks @dlech !

benwiley4000, 2021-01-30:
@jorgePe I tried what you wrote but I get this error message from `amidicat`:

```
Unable to connect from ALSA port 128:0: Operation not permitted
```

Does that sound familiar?

JorgePe, 2021-01-30:
do you have timididy on port 128:0..3?
You can list all your MIDI devices with `aconnect -l`

```
~$ TiMidity starting in ALSA server mode
Opening sequencer port: 128:0 128:1 128:2 128:3
aconnect -l
client 0: 'System' [type=kernel]
    0 'Timer           '
    1 'Announce        '
client 14: 'Midi Through' [type=kernel]
    0 'Midi Through Port-0'
client 128: 'TiMidity' [type=user,pid=291860]
    0 'TiMidity port 0 '
    1 'TiMidity port 1 '
    2 'TiMidity port 2 '
    3 'TiMidity port 3 '
```
sorry, all my EV3 nearby are in use so I am using my Ubuntu linux laptop but will release one soon

benwiley4000, 2021-01-30:
Yep aplaymidi works fine just amidicat doesn't seem to work. Not sure if I'm sending the message incorrectly though.

JorgePe, 2021-01-30:
If aplaymidi is working you for now avoid the pipe and try amidicat directly
http://krellan.com/amidicat/amidicat.html

This examples plays a note directly to Timidity port (I forgot that you can use devices names... 'TiMidity' is easier than '128:0' and it works even it assigned to another port): 
`echo "903C7F" | amidicat --port "TiMidity" --hex`

benwiley4000, 2021-01-30:
> This example plays a note directly to Timidity port (I forgot that you can

Unfortunately I get the same error when I run that. I will try to look into
alternative ways to send stuff to timidity, I'm just not sure what the
message format is.

benwiley4000, 2021-01-31:
I found a working solution! Start the timidity server:

```
timidity --output-mono --sampling-freq=25000 -iA
```

And send messages using the mido python library: https://mido.readthedocs.io/en/latest/

For example this worked for me:

```python
import mido
msg = mido.Message('note_on', note=50, velocity=80)
port = mido.open_output('TiMidity port 0')
port.send(msg)
```

JorgePe, 2021-01-31:
Uff.... I got it.
My recent MIDI incursion assume a synth OUTSIDE the EV3 so I never tested amidicat with Timidity

You need to use the 'Midi Through' port (usual 14:0) instead ot Timidity (not sure why because the examples were supposed to work... 'amidicat' is very old and something changed in ALSA or Timidity.

So:

```
sudo apt install timidity
wget https://github.com/JorgePe/randomideas/blob/master/MIDI/DrumKit/amidicat
sudo chmod +x  ./amidicat
timidity -Os -iA --output-mono --sampling-freq=22500 &
./amidicat --port 14:0 --hex < ./midipipe &
aconnect 14:0 128:0
```

Then you can send a note ON:
```
robot@ev3dev:~/miditest$ echo "90 3C 7F" | ./amidicat --port 14:0 --hex
90 3C 7F
90 3C 7F
Input total: 1 MIDI messages, 3 bytes
Output total: 1 MIDI messages, 3 bytes
```

and a note OFF:
```
robot@ev3dev:~/miditest$ echo "89 3C 00" | ./amidicat --port 14:0 --hex
89 3C 00
89 3C 00
Input total: 1 MIDI messages, 3 bytes
Output total: 1 MIDI messages, 3 bytes
```

It works but the NOTE ON decays and mute almost immediately. I guess it is because of the soundfont - same messages sent to my MiDIPLUS Engine do not decay. If I am right you will have to find a proper GM soundfont and edit "/etc/timidity/timidity.cfg" to use it instead of the default (freepats).

Also tested the pipe inside micropython (Pybricks):
```
mkfifo midi pipe
brickrun -r -- pybricks-micropython
Pybricks MicroPython v1.11 on 2020-05-06; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> pipe = open("./midipipe", "w")
>>> pipe.write("90 3C 50")
90 3C 50
8
>>> pipe.write("90 3C 00")
8
>>> 90 3C 00

```

JorgePe, 2021-01-31:
mido is powerfull... but I had problems installing it on ev3dev in 2016 - glad you manage to do it

benwiley4000, 2021-01-31:
Aha well I'm glad it works now! And thanks for all your findings, I'll check it out.

benwiley4000, 2021-01-31:
Not sure if this was your problem, but I'm guessing you may have been using
python3, which is required for the main ev3dev python package. I wasn't
able to get mido installed with pip3, but I did install the python-mido
package from the debian repositories (which works with python 2). So to
avoid having to use python3 I'm using the python 2 bindings for the
ev3dev-c library, which seem to work pretty well so far!

JorgePe, 2021-02-01:
Thanks for the explanation.
I might give it a try after soon, after I finish my MIDI instruments I want to return to the a parallel idea of reacting to MIDI messages so I can have robots that play music and also robots that are controlled by music.

JorgePe, 2021-02-02:
Long thread but still useful (at least for crazy MIDI robot makers) - got 'mido' working with python3.
Whole process takes about 1 hour or maybe a few minutes more:

```
sudo apt install python3-pip
pip3 install mido
sudo apt install pkg-config build-essential libasound2-dev python3-dev
pip3 install python-rtmidi --install-option="--no-jack"
```

I had already timidity 'libasound2-dev' and 'python3-dev' installed (and not sure if 'libasound2-dev' results from 'timidity' installation) so just 'pkg-config' and 'build-essential' were needed.
'python-rtmidi' is needed as a backend for 'mido' to access MIDI ports. Since ev3dev only uses ALSA, "--no-jack" should help reducing overhead and/or time to install.

on the last step ('python-rtmidi') I had constant crashes after an hour or so, related to memory. But adding a swapfile didn't fixed so I disabled the use of '/dev/zram0' as swapfile and used just a temporary swapfile on the microSD card (I believe this frees some precious bytes of real memory because now this last step executes in less than 30m)
```
sudo swapoff -v /dev/zram0
sudo fallocate -l 512M /swapfile
sudo chmod 600 /swapfile
sudo mkswap /swapfile
sudo swapon /swapfile
```
Also used 'ev3dev-config' to disable 'avahi-daemon' and 'brickman' to release some resources but probably not necessary for the installation (but can be useful later for intensive MIDI uses)

To disable the swapfile:
```
sudo swapoff -v /swapfile
sudo rm /swapfile
```

Ben's example worked with python3
https://github.com/ev3dev/ev3dev/issues/726#issuecomment-770312836

Thank you all. MIDI on EV3 is something that might look awkward but it is also another proof of how 'ev3dev' is a great collaboration project.

benwiley4000, 2021-02-02:
Interesting, you didn't have any errors after running pip3 install mido? Maybe it was because I had already installed the python2 version?

JorgePe, 2021-02-02:
no errors but only tried your example after succeeding with python-rtmidi.
probably not because python2 - i would bet on some of the dependencies.
In a few days I will retry from zero with a clean installation of ev3dev to document the process. Perhaps until then I also find a more suitable GM soundfont.

benwiley4000, 2021-02-02:
Yeah I would be interested to know about the best soundfont to use. I have one that I tend to use in my projects but it's too large to fit in the ev3's memory

JorgePe, 2021-02-02:
about the decaying note, it seems to be something specific to midi.Message('note_on') - will have to read docs.
This does not decay with default freepats soundfont - as expected

msg_on = mido.Message.from_bytes([0x90, 0x3C, 0x7F])
msg_off = mido.Message.from_bytes([0x90, 0x3C, 0x7F])

benwiley4000, 2021-02-02:
So now I have a program that plays a Midi note every time I press or bump the touch sensor. Cool! But a noticeable latency of several hundred milliseconds. I know this was mentioned already, but I'm wondering if anyone had luck reducing that latency?

I'm gonna try the solutions mentioned in this thread which might work: https://rosegarden-user.narkive.com/22gSf7fE/newbie-alert-huge-lag-latency-using-timidity-or-fluidsynth-w-c-media-8738

JorgePe, 2021-02-02:
Good luck!
But remember that total latency is not just only on Timidity side - the ev3dev ALSA audio card is also software based and python itself also adds some lag to it. And the EV3 ARM is not meant for audio (but you can overclock it a bit, look for David's HOWTO).
If you really really need good latency, connect a USB-MIDI adapter to your EV3 (it should show up as port 20:0) and then to a real synth.

benwiley4000, 2021-02-02:
I solved it! I had to reduce the audio buffer size for ALSA and the latency became negligible - totally acceptable by my standards. I changed the buffer size from 16834 to 2^10 = 4096. I also had to reduce the sampling rate from 25000 to 20000, which still sounds basically the same, in order to get rid of audio artifacts with the reduced buffer size. I get a warning about the buffer being too small, but at least for piano it still sounds fine.

This worked for me:

```console
$ timidity --output-mono --sampling-freq=20000 -iA -B 2,10
Requested buffer size 4096, fragment size 2048
ALSA pcm 'default' set buffer size 4096, period size 4096 bytes
Warning: Audio buffer is too small.
TiMidity starting in ALSA server mode
Opening sequencer port: 128:0 128:1 128:2 128:3
```

JorgePe, 2021-02-02:
Do you mind if I quote your findings on a post I am planning later on my blog?

benwiley4000, 2021-02-02:
Go for it!