ev3dev/ev3dev #728: Using Docker to cross compile - error while loading shared libraries: libstdc++.so.6

volansky, 2016-08-23:
Following the steps of - http://www.ev3dev.org/docs/tutorials/using-docker-to-cross-compile/ at the section of cross compiling the hello.c program got this error message: 

> /opt/gcc-linaro-arm-linux-gnueabihf-4.8-2014.04_linux/bin/arm-linux-gnueabihf-gcc: error while loading shared libraries: libstdc++.so.6: cannot open shared object file: No such file or directory

I tried installing lib32stdc++ , searched for the missing file and found it in place but when opening the docker container I couldn't find the lib32 folder so I added it to my mount command (-v usr/lib32) so now I could see the file inside my container but when running the compilation command I got the same error again..

Any ideas?

my export command: 
`CC=/opt/gcc-linaro-arm-linux-gnueabihf-4.8-2014.04_linux/bin/arm-linux-gnueabihf-gcc` 

my compilation command:
`$CC --sysroot=/ -marm -march=armv4t -mfloat-abi=soft -o hello hello.c`

*I'm using ubuntu VM (VirtualBox) on windows 10.

dlech, 2016-08-23:
The missing package is `libstdc++6:i386`. I have just pushed a new image that includes this.

> *I'm using ubuntu VM (VirtualBox) on windows 10.

Why not just run Docker for Windows? It will be much better than running Docker inside a VM.

closed 2016-08-23 by dlech

volansky, 2016-08-23:
Cant install Docker for windows on home edition and the Docker toolbox (which is suggested for home edition users) got it's own issues, eventually using VM was the easiest way.

Thanks!