ev3dev/ev3dev #729: Is it possible to run gameboy on the EV3?

mrgolfr, 2016-08-29:
I'm wondering if it is possible to run a gameboy emulator on the ev3

JorgePe, 2016-08-29:
The CPU and RAM resources probably are enough for a basic gameboy emulator, the problem is graphics support (and probably some library dependencies).
[Here] (https://github.com/ev3dev/grx) it is stated tht GRX3 supports Glib (but not SDL). [Thalia] (https://github.com/TobiasKappe/Thalia) has Glib support, you might try it. It lacks full support (color, sound) but that also means it requires less hardware resources and less libraries.

Good luck!

dlech, 2016-08-31:
Indeed, the graphics library will probably be the biggest problem. (GLib is not a graphics library - Thalia requires GTK+, which is a desktop widget library, so won't work).

The problem I found with existing Linux graphics libraries (like SDL) is that they don't support 1-bit-per-pixel. And even if you did find a graphics library that worked, there is also the problem that the EV3 screen is 178x128 and the Game Boy screen is 160x144, so the images displayed would be cut off at the top and/or bottom.

dwalton76, 2016-08-31:
Maybe use a combination of pygame and an existing gameboy emulator in python (there are several on github)?

markyi370, 2017-05-16:
It's certainly an interesting idea. I might look into this.

dwalton76, 2017-05-16:
I wonder if one of the emulators that will run on a raspberry pi would work

dlech, 2017-05-16:
Especially now that we have discovered that the EV3 can do 2bpp grayscale just like the GameBoy.

dlech, 2018-04-25:
I can now definitively say, the answer is yes.<sup>*</sup>

[image]

<sup>*</sup>May require 1337 h4x0r skillz.

dwalton76, 2018-11-21:
You should do a blog on how you got this to work :)

closed 2019-03-25 by dlech