ev3dev/ev3dev #730: Bluetooth stopped working after upgrading

Kooozer, 2016-08-29:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3-ev3dev-jessie-2015-12-30.img

* My kernel version is: 4.4.17-14-ev3dev-ev3

* My host computer is running: Linux

* I am connecting using: Bluetooth

### About my issue:

So I've been happily using ev3dev for about a week now, up until this morning when I decided to run apt update and apt upgrade. I had already done this a few times before with no issues, but this time it's decided to break the Bluetooth. On selecting 'Powered', the Bluetooth symbol doesn't appear, but the menu items do. The 'Scan now' item has been replaced with '???'. I tried just formatting it and reinstalling the img above, which worked fine up until I ran apt upgrade again.

Wifi works, haven't tried USB due a lack of a mini USB cable...

Is there a way to find out what the problem is, or should I just not upgrade it? Leaving things not updated makes me nervous...

dlech, 2016-08-31:
If you want to help find bugs and make ev3dev better, I suggest using the [latest "nightly" image](https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/). Just be prepared to report issues and wait a long time for someone to fix them if you can't figure out how to fix it yourself.

If you don't want things to break, then don't upgrade.

Also, this sound like a duplicate of #572.

Fuzzyma, 2016-09-14:
One note from my side: Its ridicolously hard, to find the right link to the nighty builds. One is broken, the other one leads to the jessy build from `2015-12-30` as latest. The link you provided is the first working link with actual images.
Imo one has to add this link to the docs at a point where everyone can find it easily. Google does not help here since it always returns invalid links as result (e.g. your linked issue contains also an invalid link)

Thanks for your great work!

WasabiFan, 2016-09-14:
@Fuzzyma Yes, the link is pretty hard to find. We are currently working on a new page with instructions for upgrading ev3dev (ev3dev/ev3dev.github.io#237) which has a more prominent link for now until we do an official release that shows up on the Getting Started page. Once those changes are reviewed and merged, there will be an easier-to-find link from ev3dev.org.

As a side note, please try not to post things in issues that aren't directly related to the original issue report; I get that this is tangentially related, but having multiple discussions like this in the same issue makes it very hard to keep track of.

dlech, 2016-10-11:
Can this issue be closed?

closed 2017-02-08 by dlech