ev3dev/ev3dev #731: Can't Download Git

karruna, 2016-09-02:
Hello Everyone,

I'm using a Mindstorm Ev3, with the ev3-ev3dev-jessie-2015-12-30.img image, my computer uses Windows 10 and I'm trying to connect with the usb cable.

I'm assembling the Print3rbot project, but I can't install properly the Git aplication bacause it says that is unable to locate the packages, I already tried many solutions that I found, but none worked. 

It always returns me:

`Err http://ev3dev.org jessie InRelease

Err http://ftp.debian.org jessie Release.gpg
  Temporary failure resolving 'ftp.debian.org'
Err http://ev3dev.org jessie Release.gpg
  Temporary failure resolving 'ev3dev.org'`

What I can do to solve it?
Thank you!

(Sorry my english is really terrible...)

mob41, 2016-09-02:
Seems you don't have a proper internet connection. You need to share your Local Area Connection to your EV3 network adapter.

mob41, 2016-09-02:
Did you followed the Getting started? You can setup your network connection with this guide, and select Windoes: [http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/](http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/)

dlech, 2016-09-04:
Duplicate of #602

closed 2016-09-04 by dlech