ev3dev/ev3dev #733: Clarify "libraries" page and promote a single option for each language

WasabiFan, 2016-09-03:
The "Libraries" page is currently confusing and needlessly complicated. We should have the largest section focus on the most common and primary options, and then have links to extra resources at the end. Preferably a heading (maybe `h2` or `h3`) per language.

cc @ndward

dlech, 2016-09-03:
Sadly, 90% of the libraries should be flagged as "not actively maintained"

The only ones that have seen activity in the last 6 months (that I know of) are
* https://github.com/ev3go/ev3dev
* https://github.com/in4lio/ev3dev-c
* ev3dev-lang-*

WasabiFan, 2016-09-14:
Do we think this is done? After the overhaul of this page that I did, I believe it is in good shape.

closed 2016-09-21 by WasabiFan

jabrena, 2016-09-21:
Hi, In the case of Java, I return to the library.

Why dont exist a release for raspberry pi 3?

Cheers

WasabiFan, 2016-09-21:
> Why dont exist a release for raspberry pi 3?

I'm not quite sure what you mean... are you referring to ev3dev releases in general? If so, it isn't related to this issue (questions like that should be asked in their own issue, although in this case it has been asked before, so you should do a search beforehand). The image for the Raspberry Pi 2 supports the Raspberry Pi 3 as well, so you use the `rpi2-` image.

jabrena, 2016-09-22:
Oki, I will test with it.

Cheers