ev3dev/ev3dev #734: Review path of a new user through documentation and revise page flow

WasabiFan, 2016-09-03:
After finding the "Getting started" page on our website, a new user could easily get lost when trying to get to the point of actually writing code. I think we need to:

- Make the links to the "Libraries" page more obvious and natural as the next step (maybe even make it a button saying "Next"?)
- Demote the low-level driver docs
  - Make it easiest to find the motor class and sensor class documentation, without having to search or jump through hoops
- Make sure that a user goes through the process of configuring networking (including SSH, see #727) before being routed to any library documentation
- Recommend the simplest networking options first. I think the simplest is "buy a Wi-Fi dongle".
- Remove links to empty "TODO" pages. These pages should still exist and we should look to get them filled, but until then we need to make sure that users don't stumble across them and get confused.
  - Make sure we aren't recommending nonexistent options
- Promote reporting issues and getting in chat (so, essentially the support page)

Essentially, I want to make sure that any user can get ev3dev running and then write some code in as few steps as possible, while also having the option to find the lower-level docs if necessary. At this point, the low-level bits are not important to the vast majority of users as far as I know. For the ones that _do_ need them, however, we need to make sure that those sections are there.

cc @ndward

Related:  #733

WasabiFan, 2016-09-14:
To get this started, I decided to take a look at the types of users I expect would be coming to our site. I essentially traced their ideal path in terms of the questions they need answered and the information they are looking for; in some cases, one "question" would make sense as one page on the site, while in other cases I think there will be 1:2 or 2:1 mappings. I included a flow chart below that illustrates this pathway as I imagine it.

Note that each class of user has a "goal" associated with it. Also keep in mind that the bolded items -- boxes, arrows, or text -- indicate something that we are missing.

[image]

From looking at that chart, it seems like we have most of the major pages in place already for the "New user" pathways, up until the "direct driver access" user merges with the other section of the chart. This also shows that here are some important parts of the "technical" flow that we lack right now. There is the "transferring code" item that we're missing.

I think that this gives us a good view of the large-scale path that users should be taking. Based on what I see here (as well as some other observations), I think that we need to modify our nav bar structure to look like this:

- Home
- Get started
- Documentation (clicking this should lead to the docs landing page)
  - Tutorials
  - Libraries
  - Low-level driver access
  - More... (also links to landing page)
	- _the landing page needs to have links to everything that is in the Documentation nav menu, at a minimum._
- Get help
- News
- Community
  - Project showcase
  - Contribute
  - Share

While I decided to omit the "download" button here, I think we might still want it there, albeit in a modified form. That will depend on if we want to break our "download" section of the getting started guide out into its own page or not. Also, something important to consider: do we want sub-items under "Documentation" at all? Maybe we should direct everyone to the main Documentation landing and then branch out from there.

Now that I've looked at the overall pathways, I also took a look at the individual pages and their content. The list of additional changes that I believe should be made are as follows. This list is a revised version of my initial thoughts above (which I have been editing as I thought of things for the last few weeks).

- Emphasize next steps in Getting Started
  - Make the links to the "Libraries" page from "Getting Started" more obvious and natural as the next step
    - This could be taken further by having the "introductory pages" display a "Next step" button at the bottom of the page. This would help with the flow of that series. Thoughts?
- Demote the low-level driver docs (partially covered above)
  - Make it easiest to find the motor class and sensor class documentation, without having to search or jump through hoops
  - Classes should be the most obvious links on the driver langind page(s)
- Make sure that a user goes through the process of configuring networking and SSH before being routed to any library documentation
  - Networking and SSH should be one of the steps in Getting Started
  - This ties into one of the points below: we need a landing page for networking.
  - This depends on a review and merge of ev3dev/ev3dev.github.io#228
- Recommend the simplest networking options first. I think the simplest is "buy a Wi-Fi dongle".
- Remove links to empty "TODO" pages. These pages should still exist and we should look to get them filled, but until then we need to make sure that users don't stumble across them and get confused.
  - Make sure we aren't recommending nonexistent options
- Promote reporting issues and getting in chat (so, essentially promote the support page)
  - I think we should have a mention of the support page on every page that a beginner sees (getting started at a minimum)
- Add more "landing pages" for categories of documentation, and make it easy to find them (especially via the documentation landing page).
  - I think that these pages would be great as a start:
    - "Connecting to the internet/a network"
	- "Installing software"
	- "Writing code using a library"
	- "Interfacing with ev3dev drivers directly".
- Explain what ev3dev is somewhere other than the homepage
- Remove "calling it a day" section
  - I think this breaks the flow of the Getting Started guide. This content is valuable, so we may want to break it out into its own page, but the last step in the guide should lead into something about writing code without any barriers in between.
- Add explanation on tutorials page about what the tutorials are
  - Maybe add descriptions of each category
- Add a link to our GitHub org
  - Somewhere in the header we should link to GitHub. It is in some ways comforting to people (from what I've seen) to know that the project they're looking at is hosted on GitHub.

---

So, what do others think? This is just a dump of my thoughts, along with some quality and roughly structured thinking time; I'd like to know if there is anything I forgot or overlooked, as well as if there are any points I made that others disagree with. In the end, my hope is to turn this into individual work items (issues) so that we can actually start getting this stuff fixed up.

WasabiFan, 2016-09-23:
@dlech I'm considering creating issues for the individual tasks from above on the site repo and adding them to a GitHub Project to both track the individual parts of this work and try out that new feature. What do you think? It will probably be something on the order of 10-20 issues.

dlech, 2016-09-23:
Go for it. It will be much easier to comment when there is one idea per issue.

WasabiFan, 2016-09-23:
Aaaaand it's live! We now have a Project tracking this work. https://github.com/ev3dev/ev3dev.github.io/projects/1

dlech, 2017-10-28:
I think some of these ideas can be applied to ev3dev-stretch docs in progress in #952