ev3dev/ev3dev #736: Docker instructions need updating for OS X for QEMU dependency

DEGoodmanWilson, 2016-09-05:
### System information:

* I am using: * LEGO MINDSTORMS EV3*

* I installed ev3dev using this image file: *ev3dev-jessie-2015-12-30/ev3-ev3dev-jessie-2015-12-30.img*

* My kernel version is: *3.16.7-ckt21-9-ev3dev-ev3*

* My host computer is running: * OS X 10.11.6*

* I am connecting using: *Bluetooth*

### About my issue:

Following the Docker tutorial, attempting to run the Docker image results in the following error:

```bash
$ docker run --rm -it -v /Users/dgoodman/Documents/Projects/ev3/:/home/compiler/example ev3cc
standard_init_linux.go:175: exec user process caused "exec format error"
```

The problem is that I skipped over the callout box labeled "Linux", because I am running OS X. So I missed out on the QEMU dependency. I am still testing all this, but it looks like the right way forward is to run
```bash
$ brew install qemu
```
(Assuming Homebrew is installed, and all that).

DEGoodmanWilson, 2016-09-05:
Update, it looks like QEMU needs to be installed with some different set of options than the default :( Will be looking into this more, would love some help from those more knowledgable than I.

dlech, 2016-09-05:
Interesting, I'm pretty sure this is working on my Mac. I'll have to double-check.

In the mean time, the key is to register the proper magic with binfmts. See http://ev3dev.org/irclog/#id-50321 for some hints.

dlech, 2016-09-05:
Not sure if the BSD kernel on Mac works the same as Linux though.

dlech, 2016-09-05:
You shouldn't need to actually install `qemu` on the Mac. It is included in the Docker image. Just registering the binfmt should be all that is needed.

dlech, 2016-09-06:
What is your docker version info? I'm running 1.12.1-beta24 (on macOS 10.12 beta7), but it should work with the stable Docker 1.12. According to this [this blog](http://blog.hypriot.com/post/first-touch-down-with-docker-for-mac/), the binfmt stuff is built-in to Docker, so if it is not working for you, then it is a Docker bug. If you are using the older Docker Toolbox for Mac, then you need to upgrade.

DEGoodmanWilson, 2016-09-08:
```bash
$ docker --version
Docker version 1.12.0, build 8eab29e
```

I'm just a bit confused—do I need to run the `binfmt` command in the IRC logs, then, or should I expect this to just work?

DEGoodmanWilson, 2016-09-08:
Also, since I can't launch the Docker container, how would I run `binfmt` in a way that it helpful?

dlech, 2016-09-08:
On OS X, it should "just work". I would suggest reinstalling docker, trying the latest docker beta or posting on the Docker forums to see if someone who knows more about Docker can help.

DEGoodmanWilson, 2016-09-08:
I can try, but this is a fresh install of Docker—I installed it just for this project—so i don't have especially high hopes in that regard.

dlech, 2016-10-11:
I have just made a new image for cross compiling that does not rely on qemu: `ev3dev/debian-jessie-cross`

Updated docs at http://www.ev3dev.org/docs/tutorials/using-docker-to-cross-compile/

Give it a try and open a new issue if you have problems with the new image.

closed 2016-10-11 by dlech