ev3dev/ev3dev #737: Have Brickman guess file type using file extension

WasabiFan, 2016-09-05:
This was suggested by @ndward in rhempel/ev3dev-lang-python#194.

The idea here is to have Brickman guess the type of a file from a set of common types and suggest adding a shebang if there isn't one already (i.e. only attempt this after execution has failed). I could imagine this being taken even further, and instead of suggesting a shebang it could just add one itself or execute the file using the normal runtime (e.g. execute the contents of the expected shebang even if there isn't one). This would make the process easier for beginners, but could also lead to confusion if it does something unexpected. Thoughts?

Additionally, something similar could be done with the executable bit: if a file that you try to execute isn't marked as such, Brickman could warn you and maybe let you "Click 'yes' to mark the file as executable". Again, this would help new users but could also lead to confusion.

JorgePe, 2016-09-07:
I think the shebang is less important than the executable bit as beginners will find out about shebangs in most examples they read (well, I did).
But as a sysadmin, giving execution permissions to files made by beginners makes me nervous :) But I guess it is OK, as long as "robot" doesn't belong to "root" group.

WasabiFan, 2016-09-07:
> I think the shebang is less important than the executable bit as beginners will find out about shebangs in most examples they read (well, I did).

That seems right. I think there are still cases where the shebang helper logic would be useful, but it's less obscure than setting the executable bit.

> But as a sysadmin, giving execution permissions to files made by beginners makes me nervous :) But I guess it is OK, as long as "robot" doesn't belong to "root" group.

Agreed. In general, I view an installation of ev3dev as disposable; if I mess something up I just re-flash the card. So, in that sense, if a beginner runs code that could damage the installation, that isn't a big issue in my mind. There _is_ the potential problem of this functionality making it easier to get malware installed by accident which could watch the network the brick is connected to. But I think that the risk is minor and the chance of getting files like that on an EV3 in the first place are slim as well.

ndward, 2016-09-08:
I think my original message was about the executable bit rather than the
shebang. Isn't there a problem with the shebang? Don't both python 2 and
python 3 scripts end with .py but don't they need different shebangs?
Offering to set the executable instead of just giving an error would surely
be helpful though.

WasabiFan, 2016-09-08:
> I think my original message was about the executable bit rather than the
shebang. 

Yes, that's one of the main points here; however I focused the issue on the other point (which is closely related) because it seemed to be the more potentially problematic. Either way, yes, that is a focus of this issue.

> Isn't there a problem with the shebang? Don't both python 2 and
python 3 scripts end with .py but don't they need different shebangs?

Yes, that is something that would need to be considered. I imagine it would make sense to assume Python 3, and tell the user that that's what was going to be used.

WasabiFan, 2016-11-15:
I'm taking a look at the tangible issues mentioned here right now to figure out how we might want to implement this. My basic idea is to have Brickman display some variety of dialog when an invalid file is selected. This means that it would have to identify the existence of shebangs and other properties (e.g. whether the file is a binary). My goal here is to make running programs accessible to beginners, so I'm targeting this at the most basic level under the assumption that more advanced users will be able to handle themselves using the more simplistic features.

My primary idea, which is fairly straightforward, is something along the lines of the following (for when no shebang is present):

[image]

Because the EV3 screen is so small, we can't give a reasonable explanation in-app. Giving a link to the site seems like the next-best option.

A similar approach could be taken with the executable bit:

[image]

Alternatively, we could allow the user to set the bit from the dialog. This is something that would persist even after altering the file with remote tools, so it would make sense to not block the user on it. I could imagine a UI such as the following:

[image]

From past experience, I believe that this text would overflow and require a scroll bar on the EV3 display, so we should take that into consideration.

Going back to the issue of shebangs, we could do something like what I showed above for the executable bit with an additional dialog and button:

[image]

---

Looking at the available options, I think I like the 2nd "executable bit" design the most out of those two; it's simple yet allows you to bypass the issue cleanly if you know what you want to do. For the issue of shebangs, I'm a bit more divided. On one hand, keeping it simple is good; on the other, allowing the user to get past issues without forcing them out of their way is also helpful. I think I would support the more complicated one with a list of known languages, with the condition that selecting a language would not actually modify the file; we would manually execute it with the selected interpreter. The user would need to follow instructions presented on our site at the provided URL to make it permanent with their editor.

What do others think? What seems most natural?

ndward, 2016-11-15:
Hi
As you know, I'm very much in favor of being able to set the executable bit
from within a dialog. Your message could be shortened to 'This file is not
marked 'executable' and therefore cannot be run (see
ev3dev.org/exec-bit-help). Make it executable and run it?'
or made even shorter if necessary:
This file is not marked executable. Make executable and run?'
or even shorter!
'File not executable! Make executable and run'
My final offer:
'Make executable and run?'

You could have three buttons: Yes, No, Help with the Help button opening a
textual message on the EV3 screen. I estimate that using the same (very
legible) font that Brickman uses, the screen can display about 200
characters (8*25) which surely should be enough to display a rather
detailed text message - it could be made scrollbale or multi-page if
necessary.

Nigel

ndward, 2016-11-15:
I'm not familiar with LeJos but I guess you and/or other EV3Dev gurus are.
Surely many of the issues being faced by EV3dev have already been faced and
solved by LeJos? Can LeJos provide inspiration for the EV3dev project?

Nigel

WasabiFan, 2016-11-15:
> You could have three buttons: Yes, No, Help with the Help button opening a textual message on the EV3 screen.

I mocked this out on-brick to see how it would fit. Not only were the buttons squished, but the space for text was so small that I didn't feel we could explain adequately. I like the idea, but I figured that a URL would work better. Do you think the advantage of it on-brick is worth the limited space?

dlech, 2016-11-15:
Instead of all of these pop-up dialogs for different things, what if we have a "properties" page for each file? Pressing enter on a file will open the properties page.

The properties page might look like this...

```
myfile.py (1k)
-----------------------
Run
Delete
Executable [X]
CR/LF > LF
```

`Run` will only appear if the file is executable (or perhaps it has a known file extension like `.py` or `.sh` - in which case, we should not give an option, just use `python3` for `.py` and `bash` for `.sh`, etc).

This could also be extended, for example, to show `Play` for `.wav` files, `View` for `.png` and so on.

`CR/LF > LF` is to fix line endings. It should only be shown for files starting with a shebang.

WasabiFan, 2016-11-15:
> Instead of all of these pop-up dialogs for different things, what if we have a "properties" page for each file? Pressing enter on a file will open the properties page.

My initial reaction to this was that it would be an extra step for little benefit which steered me away, but I think the reasoning you provided makes it a lot more appealing.

> The properties page might look like this...

I like that design overall. The one extra thing we should consider is that we need to make it clear that you're in a file properties page as opposed to a directory. Maybe something like an icon corresponding to the file type or an animation would help with that.

Am I correct that the "Executable" option is a checkbox? If so, we're going to need a way to make it clear that one needs to check that box to make the "run" button appear. This is tough on the monochromatic binary display. We could at least group "Run" and "Executable", and potentially have the "Executable" text changed to "Executable (required to run)" or similar when the file _isn't_ runnable. What do you think?

> Run will only appear if the file is executable (or perhaps it has a known file extension like .py or .sh - in which case, we should not give an option, just use python3 for .py and bash for .sh, etc).

I like that. I'd propose logic similar to the following:
- If the file is executable
  - If the file is binary, enable the button
  - If the file has a shebang, enable the button
  - Otherwise, still enable the button but indicate an unknown type
- If the file has a known extension, enable the button
- Otherwise, disable

My reasoning here is that if it's a system-recognized executable such that you could run it from a shell, we should allow running it from our UI as well. Past that we can guess your intention and help you out even if the system wouldn't run it otherwise.

We could also show information about how we're going to run it in the UI; maybe in the "run" button text (e.g. `Run (Python 3)`) or maybe in the title.

---

I think we also still need explanation somewhere about what these options mean. Maybe we could have a website URl with detailed explanations somewhere in this UI; what do you think?

dlech, 2016-11-15:
> Am I correct that the "Executable" option is a checkbox?

yes

> We could at least group "Run" and "Executable", and potentially have the "Executable" text changed to "Executable (required to run)" or similar when the file isn't runnable. What do you think

I think if we run files based on file extension when the executable bit is not set, then we don't need the executable checkbox at all. Same with the "fix the line endings" option - it's not really needed. Or, if you think we should keep them, maybe make a second page for "Advanced" options.

> Maybe we could have a website URl with detailed explanations somewhere in this UI; what do you think?

I don't think very many people would bother reading large amounts of text on the EV3 or transcribing a url on the EV3 to their web browser. I think we should just make the (yet to to be written) Brickman manual easy to find on the ev3dev.org website. If a person can't use Brickman without reading the detailed descriptions, the we are "doing it wrong".

WasabiFan, 2016-11-18:
> I think if we run files based on file extension when the executable bit is not set, then we don't need the executable checkbox at all. Same with the "fix the line endings" option - it's not really needed.

I think that makes sense. It would be nice to have these options in a submenu, but not required.

However, if we don't have these options, we will need to do a few things:
- Directly execute binaries and bypass the executable bit
- Manually execute _all other files_, parsing the shebang if it is present and choosing our own otherwise.

> I don't think very many people would bother reading large amounts of text on the EV3 or transcribing a url on the EV3 to their web browser. I think we should just make the (yet to to be written) Brickman manual easy to find on the ev3dev.org website. If a person can't use Brickman without reading the detailed descriptions, then we are "doing it wrong".

I agree, but we need to make sure that we aren't leaving people confused in the interim. I'd say that if we don't have a usable "Brickman manual" by the time that our approximation of ev3dev v1.0 ships we'll need to find an alternative to include. It would also be nice to have a first-boot dialog that advertised this manual.