ev3dev/ev3dev #739: ev3dev Won't Boot, MicroSD Card in Mindstorms EV3

webmanoffesto, 2016-09-09:
I have a new Lego Mindstorms brick and I want to run ev3dev on it, mainly because all my computers are Linux.

I followed the instructions at http://www.ev3dev.org/docs/getting-started/ but ev3dev won't boot.

I used Etcher https://www.etcher.io/, like the instructions say, and it said that I successfully flashed the image to the MicroSD card. The card is 2GB, not oversized.

On the SD card I see the EV3DEV_BOOT and EV3DEV_ROOT partitions.

The lights do not change from red to orange/amber or flash when booting.

I do have a a serial cable which I can use for debugging. It's the standard "LEGO Cable USB to mini-usb for Mindstorms EV3".

WasabiFan, 2016-09-09:
Do you have anything plugged into the brick (dongles, motors, sensors, etc.)? If so, unplug them; that might help. Also, make sure you are running on full batteries or if you're using a rechargeable that it is plugged in.

dlech, 2016-09-09:
Please answer the questions in the issue template:

https://github.com/ev3dev/ev3dev/blob/master/ISSUE_TEMPLATE.md

You probably installed the wrong image, but since you did not answer the questions, we don't know what you actually did.

webmanoffesto, 2016-09-12:
Sorry for the bad forum etiquette. Here is a more correctly structured forum question.

System information:
- **I am  using**: Lego EV3 brick
- **I installed ev3dev using this image file**: evb-ev3dev-jessie-2016-01-25.img
Flashed using Etcher-linux-x64.AppImage
- **My kernel version is**: I don't understand. I guess it's whatever the evb-ev3dev-jessie-2016-01-25.img uses
- **My host computer is**: an Asus K55A laptop running Ubuntu 16.04 LTS with Gnome desktop
- **I am connecting using**: the standard USB to mini USB cable which came with the brick

**About my issue**:

I have a new Lego Mindstorms brick and I want to run ev3dev on it, mainly because all my computers are Linux.

I followed the instructions at http://www.ev3dev.org/docs/getting-started/

But ev3dev won't boot.

I used Etcher https://www.etcher.io/, like the instructions say, and it said that I successfully flashed the image to the MicroSD card. The card is 2GB, not oversized.

On the SD card I see the EV3DEV_BOOT and EV3DEV_ROOT partitions.

The lights do not change from red to orange/amber or flash when booting.

I do have a a serial cable which I can use for debugging. It's the standard "LEGO Cable USB to mini-usb for Mindstorms EV3".

WasabiFan, 2016-09-12:
There's your problem. You are using the image for the EVB, which isn't the same as the EV3. You should flash the image corresponding to your platform.

Where did you download this image from? If you made this mistake, chances are that we need to clarify something.

webmanoffesto, 2016-09-12:
Thank you, I'll try installing the correct image later.

I got it here https://github.com/ev3dev/ev3dev/releases

Is this the correct image?
rpi-ev3dev-jessie-2015-09-15

WasabiFan, 2016-09-12:
No, that is not the correct one. Look at the prefix of the image name; this one says `rpi-`, so it is for the Raspberry Pi. To find the right image, use the download buttons half way down the getting started page: http://www.ev3dev.org/docs/getting-started/. Find the one that says LEGO MINDTSORMS EV3. Why did you decide to go to the GitHub releases page instead of using those buttons?

webmanoffesto, 2016-09-12:
Third Flash is the charm. I got it. Thank you for the help!
[image]

WasabiFan, 2016-09-12:
Nice! Glad it worked.

Remember to close the issue if your problem has been solved.

closed 2016-09-13 by webmanoffesto