ev3dev/ev3dev #741: Restructure nightly build directory structure to put the platform first

WasabiFan, 2016-09-11:
In most cases, we want to tell someone to go download a nightly release for a specific platform. For that reason, I think that it would make more sense to have a folder structure in the nightly image repository similar to this (assuming it is possible, which it seems to be):

- `brickstrap/`
  - `ev3/`
    - `nightly-ev3dev-jessie-ev3-generic-2016-08-20.img.xz`
    - `nightly-ev3dev-jessie-ev3-generic-2016-08-30.img.xz`
    - `nightly-ev3dev-jessie-ev3-generic-2016-09-04.img.xz`
    - ...
  - `rpi/`
    - `nightly-ev3dev-jessie-rpi-generic-2016-08-20.img.xz`
    - `nightly-ev3dev-jessie-rpi-generic-2016-08-30.img.xz`
    - `nightly-ev3dev-jessie-rpi-generic-2016-09-04.img.xz`
    - ...
  - `rpi2/`
    - ...
  - `bone/`
    - ...

With this approach, if we are in a discussion with someone in an issue, we can link them directly to a list of image files that they can use. They just need to scroll to the bottom and click the link. If we are unsure of the platform, we can say something like this:

> Go [here]() if you are using a LEGO MINDSTORMS EV3, [here]() if you are using a Raspberry Pi Zero / 1, etc.

This makes it simpler for us and simpler for someone attempting to follow our instructions.

dlech, 2016-09-11:
It would be even simpler to just give an absolute link to the file. Then directory structure doesn't matter. 

Personally, I find having them grouped by release (date) helpful because I can quickly see how many people have downloaded each release.

WasabiFan, 2016-09-11:
> It would be even simpler to just give an absolute link to the file. Then directory structure doesn't matter. 

I agree, however unfortunately this doesn't work in practice for most situations. In a tutorial, we need to be able to link to the whole list so that they can choose to get the most recent version or go back to a specific release if they have reason to do so. In issue comments, the point is to document the problem so that others can also follow the instructions in the future; if we linked to a specific release, someone looking at an issue a month after the nightly link was posted would be unable to choose a more recent release. So this does work well when we're telling someone "go download this specific version," but in most cases we aren't doing that.

> Personally, I find having them grouped by release (date) helpful because I can quickly see how many people have downloaded each release.

Fair point. I think we need to prioritize ease-of-use for a user over our ability to see download numbers, though. This structure would shorten the relevant upgrade instructions by a factor of two _and_ eliminate the need for a user to look through a list of names to pick out relatively minor differences (they just need to click the link that we tell them to and click the last item on that page). For that kind of an improvement, us having to check four separate places when we want to see download numbers is pretty minor.

dlech, 2016-10-11:
I'm calling this `wontfix` because I already have the process of building and uploading nightly images fully automated and rearranging the existing images would not be trivial. It doesn't seem worth the effort to make the change at this point.

closed 2016-10-11 by dlech

WasabiFan, 2016-10-11:
That's unfortunate, but obviously whatever you choose is what we're going with. This means we will need to continue maintaining the extra paragraphs of explanation + table on the site and include similar explanations whenever we link people to the nightlies.