ev3dev/ev3dev #744: Some motor driver docs need polishing/clarification

kortschak, 2016-09-18:
While writing tests for ev3go driver handlers, I noticed some minor deficits in the driver docs for the motors. I want to check here before I send a PR fixing these.

1. The servo-motor docs do not mention a uevent attribute, but it appears to be present in the source code.
2. There is variable mention of whether negative durations are valid - some attributes say so, others don't. If it is alway invalid to have a negative duration then probably it should be said so in all places.

dlech, 2016-09-18:
1. the `uevent` attribute is not specific to any ev3dev drivers. It is a core part of sysfs and is used by udev and is present on all device nodes in sysfs. In the case of ev3dev drivers, all information contained in `uevent` is duplicated in other attributes, so there is no need to read it.

2. A negative duration (time) is never valid. Perhaps you mean something else?

kortschak, 2016-09-18:
Great, those those clear up my questions.

I figured that uevent was a generic attribute, it's just that it's not mentioned in some drivers- for clarity it should be consistent. Similarly the invalidity of negative durations is mentioned in spike places but not others - the inconsistency is confusing. I plan to send a PR to add the missing mentions.

dlech, 2016-10-11:
Can you point out where we mention `uevent`?

dlech, 2016-10-11:
> Similarly the invalidity of negative durations is mentioned in some places but not others - the inconsistency is confusing. I plan to send a PR to add the missing mentions.

I believe you have done this already, correct?

kortschak, 2016-10-11:
Not yet. Things have got in the way - hopefully this weekend.

kortschak, 2016-10-13:
>Can you point out where we mention `uevent`?

I am beginning to believe I confabulated that.

closed 2016-10-15 by dlech