ev3dev/ev3dev #745: decision help needed - brickpi or pistorms

marev3, 2016-09-18:
I already own an ev3 educational set but I want to change the platform to a raspberry pi because I want to build a robot which uses the episode cam with its csi connector.

So which should I choose pistorms or brickpi?
What are your experiences?

jabrena, 2016-09-18:
Both hats are good. It is true that exist more people with BrickPi here but it is true that it is a product which it was released before. PiStorms has a Screen and it is easy to handle Brickman with a Keyboard connected. Both are good products, I have both.

Juan Antonio

marev3, 2016-09-19:
Hi Juan,

thanks for your reply. 

The brickpi has gpio pins on the top side. Can I  use these like the ones directly on the raspberry pi, i.e. connect non-lego-sensors on the brickpi? Does the pistorms also have this possibility? 

How do you rate the power consumption of both hats?

There's a 6xAA adapter available for the pistorms. How do I connect this?

Many questions,  I know...

Kind regards

Martin

dlech, 2016-09-19:
> The brickpi has gpio pins on the top side. Can I use these like the ones directly on the raspberry pi, i.e. connect non-lego-sensors on the brickpi?

Yes. Some of the pins are used by BrickPi, so you can't use these, but you can use the other pins. However, ev3dev is not based on the official RPi Foundation Raspbian, so things like WiringPi will probably not work (unless you are knowledgeable about embedded Linux systems and know how it all works under the hood).

> Does the pistorms also have this possibility?

If you have an RPi with the 40-pin connector (B+, 2, 3, etc), then pins 27 through 40 are available. It does not have pass-through for the other pins.

> How do you rate the power consumption of both hats?

I haven't actually measured, but I would expect the PiStorms to use about 80-100mA more because of the backlight on the display.

> There's a 6xAA adapter available for the pistorms. How do I connect this?

http://www.mindsensors.com/blog/how-to/pistorms-frame-assembly

WasabiFan, 2016-09-19:
@dlech Are there differences in functionality? I recall there being caveats with both devices on various modes of motor feedback, certain port functionality, etc.

We should write this up as a documentation page.

bmegli, 2016-10-12:
See #757 for I2C devices that currently don't work on BrickPi+ or BrickPi regular ports.
(I don't know how it is with PiStorms)

closed 2017-10-03 by dlech