ev3dev/ev3dev #746: Can the rpi2-ev3dev-jessie-2015-12-30.img image be used to boot RPi3 with BrickPi+?

mike-shock, 2016-09-23:
* I am using: 
   Rapsberry Pi Model 3 B
   Dexter Industries BrickPi+ (HW v2.8)

* I installed ev3dev using this image file: rpi2-ev3dev-jessie-2015-12-30.img

* My host computer is running: Linux (SUSE Tumbleeweed)

* I am connecting using: USB keyboard & mouse, TVset via HDMI.

 * About my issue:
I tried to boot RPi from several different microSD cards with the same result: rainbow square on the screen and the green LED flashing 7 times.

dlech, 2016-09-23:
This image does not work with RPi 3. Please use the latest nightly image from https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/

Also be sure to read `config.txt` carefully and edit it to match the hardware that you have.

dlech, 2016-09-23:
To be more specific, pick the image file with `-rpi2-` in the name.

dlech, 2016-10-11:
Please open a new issue if the nightly images don't work for you.

closed 2016-10-11 by dlech