ev3dev/ev3dev #747: This works for RPi OTG - Yes! - Raspberry Pi people!

putztzu, 2016-09-24:
This is an FYI -
This "obsolete" configuration works to deploy and set up a RPi OTG as described at
https://gist.github.com/gbaman/975e2db164b3ca2b51ae11e45e8fd40a

Compared to the "new, improved" guide, I don't know if configuring the first, main network is better, it's definitely not desired when the first network connection is WiFi in a public hotspot where I am writing this now.

After slamming my head against a wall for hours trying to set up Avahi (on a Linux OS), then Bonjour and iTunes on a Windows 10 as suggested in the above link original instructions, I then turned my attention to looking for ways to set up a manual network over USB connection and happened on this cookbook. 

Had to reboot the Windows machine to clear its failed attempts, only then these steps worked like a charm connecting from a Windows 10 to a RPi Zero running raspbian lite. I have little doubt this should work on the full raspbian as well, and intend to try other distros.

Very cool, and a big thank you to the author of this piece.

dlech, 2016-10-11:
Need to look into this some more. We could create a systemd service like we do for usb gadget on EV3. Then only adding the line `dtoverlay=dwc2` to `config.txt` would be needed.

Also see #721.

dlech, 2017-08-12:
Not sure if there is anything actionable here.

closed 2017-08-12 by dlech