ev3dev/ev3dev #748: "make install" fails, c library, permissions issue.

towtruck1, 2016-09-24:
Hello!

I am trying to get up-and-running using c. I think I have gone through all steps in the various tutorials successfully, but now I am getting hung up trying to install the c libraries. 

image: ev3-ev3dev-jessie-2015-12-30.img.xz

kernel: 3.16.7-ckt21-9-ev3dev-ev3 #1 PREEMPT Tue Dec 15 15:16:17 CST 20153.16.7-ckt21-9-ev3dev-ev3 #1 PREEMPT Tue Dec 15 15:16:17 CST 2015

library: ev3dev-c

The previous 'make' command appears successful. Now when I try 'make install', I get:

```
robot@ev3dev:~/ev3dev-c/source/ev3$ make install
cp -f  ../../lib/./libev3dev-c.a /usr/local/lib
cp: cannot create regular file ‘/usr/local/lib/libev3dev-c.a’: Permission denied
makefile:274: recipe for target 'install' failed
make: *** [install] Error 1
robot@ev3dev:~/ev3dev-c/source/ev3$
```

I looked into the directory structure and noticed that the permissions looked a bit different for directory 'local', but I didn't want to attempt any changes at this point.

```
robot@ev3dev:/usr$ ls -lsa
total 60
 4 drwxr-xr-x 10 root root   4096 Dec 30  2015 .
 4 drwxr-xr-x 21 root root   4096 Jan  1  2000 ..
24 drwxr-xr-x  2 root root  24576 Sep 24 20:08 bin
 4 drwxr-xr-x  2 root root   4096 May  1  2015 games
 4 drwxr-xr-x 33 root root   4096 Sep 24 20:06 include
 4 drwxr-xr-x 41 root root   4096 Sep 24 20:02 lib
 4 drwxrwsr-x 10 root staff  4096 Dec 30  2015 local
 4 drwxr-xr-x  2 root root   4096 Dec 30  2015 sbin
 4 drwxr-xr-x 84 root root   4096 Sep 24 20:07 share
 4 drwxr-xr-x  2 root root   4096 May  1  2015 src
robot@ev3dev:/usr$
```
 Thanks in advance of for any help!

dwalton76, 2016-09-24:
Try "sudo make install"

WasabiFan, 2016-09-24:
This is not an issue with ev3dev; this should be opened on the library repository, not here.

towtruck1, 2016-09-24:
Color me sheepish. That was obviously it. Thanks!

towtruck1, 2016-09-24:
Color me sheepish. "sudo make install" was obviously the answer. Thanks dwalton76!

closed 2016-09-24 by towtruck1