ev3dev/ev3dev #749: Write "upgrading guide" for upgrade to recent kernels

WasabiFan, 2016-09-25:
With an expected release looming, we're going to need to consider how we communicate the changes that one must make to their code to upgrade. This content could go in the release notes/news item or it could be its own page on the site. Mainly, I think we need to list the breaking changes and explain how to use speed instead of duty cycle.

WasabiFan, 2016-09-27:
Where would it make the most sense to put this? In the release announcement, as a tutorial, as a static docs page, or something else?

dlech, 2016-09-27:
The breaking changes have already been listed in the kernel release note. For code changes, it is up to individual libraries to list that in their own documentation. I will mention the motor changes again in the release notes.

WasabiFan, 2016-09-27:
> I will mention the motor changes again in the release notes.

OK, make sure you explain the _how_ of the new interface as well as the _what_; we need to make sure we minimize the number of people opening issues to complain about their code not working, because I expect the number of people that may be confused is pretty large. The individual libraries should document it, but honestly I doubt that people read the release notes there anyway.

Also, given that we will be releasing this one on GitHub well, it would be best to include a sentence or two in the release description about this along with a link back to the news article and release notes. people going from the Getting Started guide will see the text in the release, but not necessarily that in the news or release notes.

This is exciting 😀

dlech, 2016-11-17:
Release has been out for a month, so I don't think there is anything else to do here. Plus, there is <http://www.ev3dev.org/docs/tutorials/upgrading-ev3dev/>.

closed 2016-11-17 by dlech