ev3dev/ev3dev #750: python code in c libraries; use of swig etc.

towtruck1, 2016-09-27:
Hello

I am hoping someone can explain the relationship between the python code & C code in the EV3 C libraries. I understand from reading up on SWIG, that C/C++ code can be wrapped for use in scripting languages, however it seems at some point that the opposite is occurring. To wit:

In trying to understand the example (ev3dev-c\eg\tacho\tacho.c) I tried to discover where DESC_LIMIT was declared/defined/set.

dlech, 2016-09-27:
You might have better luck asking at the source of the library: https://github.com/in4lio/ev3dev-c/issues

towtruck1, 2016-09-27:
Thanks. Sorry I missed that. I have moved it to: <https://github.com/in4lio/ev3dev-c/issues/3>

closed 2016-09-27 by towtruck1