ev3dev/ev3dev #751: Is it possible to include ttf-mscorefonts-installer package into the image file?

ddemidov, 2016-09-27:
See https://github.com/rhempel/ev3dev-lang-python/issues/194#issuecomment-249662989 for discussion.

ddemidov, 2016-09-27:
Pinging @ndward

dlech, 2016-09-27:
We can. I think bitmap fonts look much better on the EV3 than TTFs though.

http://stackoverflow.com/questions/12384838/using-fonts-in-pil-without-freetype#12388342

dlech, 2016-09-27:
Ignore the part in the stack overflow link about converting ttf fonts. I just included it to give a hint on how PIL uses bitmap fonts.

dlech, 2016-09-27:
Perhaps ev3dev-lang-python could use this tool to create some "built-in" fonts that look really good on the EV3.

ddemidov, 2016-09-27:
The biggest problem with this (at least for me) would be to choose what fonts and at what size to include into the library. But I agree having some defaults with a nice interface around them would be nice. @ndward, you played with TTF fonts on EV3, do you have any preference by now?

ndward, 2016-09-29:
I only tried a couple of standard fonts: Arial, Courier_New and Verdana (Verdana didn't work). Courier_New is less attractive but has the big advantage of being fixed-width. Some examples (there are some odd characters included to test descenders and tiny artefacts caused by my lazy way of running scripts from SSH). It's a pity `multiline` and `align` aren't available.

Arial 22:
[image]

Courier_New_Bold 22:
[image]

Arial 38:
[image]

Arial_Bold 38:
[image]

Courier_New_Bold 32:
[image]

Courier_New_Bold 42:
[image]

(Courier_New is too skinny.)
Here is the script I used:
```
#!/usr/bin/env python3
from ev3dev.ev3 import *
from PIL import Image, ImageDraw, ImageFont  
from time import sleep

lcd = Screen()

f = ImageFont.truetype('/usr/share/fonts/truetype/msttcorefonts/Courier_New_Bold.ttf', 42)
lcd.draw.text((2,0),  'This is_py', font=f)
lcd.draw.text((2,40), 'Couriyr', font=f)
lcd.draw.text((2,80), 'size_42 yh', font=f)
'''
f = ImageFont.truetype('/usr/share/fonts/truetype/msttcorefonts/Arial_Bold.ttf', 38)
lcd.draw.text((2,0),  'This is_py', font=f)
lcd.draw.text((2,40), 'Arial_Bold', font=f)
lcd.draw.text((2,80), 'size 38 yh', font=f)

f = ImageFont.truetype('/usr/share/fonts/truetype/msttcorefonts/Courier_New_Bold.ttf', 32)
lcd.draw.text((2,0),  'Courier', font=f)
lcd.draw.text((2,31), '_New_Bold', font=f)
lcd.draw.text((2,62), 'font yhph', font=f)
lcd.draw.text((2,93), 'size 32 y', font=f)
'''
lcd.update()

sleep(12)  # for Brickman
```

ndward, 2016-09-29:
> The biggest problem with this (at least for me) would be to choose what fonts and at what size to include into the library.

But aren't TrueType fonts scalable so one does not have to choose a size?

ddemidov, 2016-09-29:
My understanding is one needs to chose the exact size to render a truetype font into a bitmap (rasterized) font.

dlech, 2016-09-29:
Truetype fonts will still look bad at sizes < 20 even if converted to a bitmap. I am suggesting that you use actual bitmap fonts that were created for fixed sizes. For example, most, if not all of the `xfonts-*` packages are bitmap fonts. `xfonts-75dpi` would be a good place to start. These packages ship the fonts in a .pcf format. However, there is a `pcf2bdf` tool to "decompile" them. to .bdf. Then use the info in the stackoverflow link above to convert the .bdf PIL font format.

If you make this part of the build process for the ev3dev-lang-python project, then you can just ship the PIL format fonts and have good looking easy to use fonts.

ddemidov, 2016-09-30:
See ev3dev/docker-library#2

closed 2016-09-30 by ddemidov