ev3dev/ev3dev #752: Motor appears to stop responding after running at high speeds

WasabiFan, 2016-09-27:
As reported in rhempel/ev3dev-lang-python#220.

It seems that, if you run a motor multiple times at high speed, it stops responding to commands. This can be reproduced with the following script (replacing `motor2` with an appropriate name):

```bash
motor=/sys/class/tacho-motor/motor2

echo 5000 > $motor/time_sp
echo 900 > $motor/speed_sp
echo run-timed > $motor/command

sleep 7

echo 5000 > $motor/time_sp
echo 900 > $motor/speed_sp
echo run-timed > $motor/command

sleep 7

echo 5000 > $motor/time_sp
echo 1000 > $motor/speed_sp
echo run-timed > $motor/command

sleep 7

echo 5000 > $motor/time_sp
echo 1000 > $motor/speed_sp
echo run-timed > $motor/command

```

You'll see that the motor runs the first three times, but the fourth it does nothing. The first two runs are with a `speed_sp` of `900` while the latter two are with `1000`. I tested this with both an EV3 and an NXT motor.

dlech, 2016-11-14:
Moving [comment](https://github.com/ev3dev/lego-linux-drivers/commit/6ad31886d442530ebd4ed812fb451ee0f19b1731#commitcomment-19816641) here to make it easier to find later...

@WasabiFan said...

> How do you debug these things? Without debugging tools or any knowledge of why these things are happening, I'd imagine it would be extremely difficult... Is it just a matter of intuiting the vague location and then adding print lines and hoping?

Yes, bugs like this can be quite tricky. I really had no idea where to start with this bug because there was nothing obviously wrong with the code. And so, yes, I started by adding some `printk()`s in some related areas to confirm that the code was following the path that it was designed to. This didn't turn up anything useful though.

Recently I added debugfs entries to the EV3 motor driver, so I decided to give them a try here and see what happened. I made a script from the code in the original comment here that ended up looking like this...

```
root@ev3dev:~# cat issue752.sh 

motor=/sys/class/tacho-motor/motor0

function debug()
{
    for f in /sys/kernel/debug/outA/*; do
        echo -n "$f: "
        cat $f
    done
}

echo run
echo 5000 > $motor/time_sp
echo 900 > $motor/speed_sp
echo run-timed > $motor/command
debug
sleep 5

echo run
echo 5000 > $motor/time_sp
echo 900 > $motor/speed_sp
echo run-timed > $motor/command
debug
sleep 5

echo run
echo 5000 > $motor/time_sp
echo 1000 > $motor/speed_sp
echo run-timed > $motor/command
debug
sleep 5

echo run
echo 5000 > $motor/time_sp
echo 1000 > $motor/speed_sp
echo run-timed > $motor/command
debug
sleep 5
```

Which produced the following output:

```
root@ev3dev:~# bash issue752.sh 
run
/sys/kernel/debug/outA/dir_chg_samples: 127
/sys/kernel/debug/outA/duty_cycle: 99
/sys/kernel/debug/outA/got_new_sample: N
/sys/kernel/debug/outA/hold_pid_ena: N
/sys/kernel/debug/outA/max_us_per_sample: 100000
/sys/kernel/debug/outA/num_samples: 16
/sys/kernel/debug/outA/position: 26532
/sys/kernel/debug/outA/position_sp: 0
/sys/kernel/debug/outA/ramping: N
/sys/kernel/debug/outA/run_direction: 1
/sys/kernel/debug/outA/run_to_pos_active: N
/sys/kernel/debug/outA/run_to_pos_stop_action: 0
/sys/kernel/debug/outA/speed: 900
/sys/kernel/debug/outA/speed_pid_ena: Y
/sys/kernel/debug/outA/stalled: N
/sys/kernel/debug/outA/state: 0
run
/sys/kernel/debug/outA/dir_chg_samples: 122
/sys/kernel/debug/outA/duty_cycle: 99
/sys/kernel/debug/outA/got_new_sample: Y
/sys/kernel/debug/outA/hold_pid_ena: N
/sys/kernel/debug/outA/max_us_per_sample: 100000
/sys/kernel/debug/outA/num_samples: 16
/sys/kernel/debug/outA/position: 31472
/sys/kernel/debug/outA/position_sp: 0
/sys/kernel/debug/outA/ramping: N
/sys/kernel/debug/outA/run_direction: 1
/sys/kernel/debug/outA/run_to_pos_active: N
/sys/kernel/debug/outA/run_to_pos_stop_action: 0
/sys/kernel/debug/outA/speed: 901
/sys/kernel/debug/outA/speed_pid_ena: Y
/sys/kernel/debug/outA/stalled: N
/sys/kernel/debug/outA/state: 0
run
/sys/kernel/debug/outA/dir_chg_samples: 0
/sys/kernel/debug/outA/duty_cycle: 100
/sys/kernel/debug/outA/got_new_sample: N
/sys/kernel/debug/outA/hold_pid_ena: N
/sys/kernel/debug/outA/max_us_per_sample: 100000
/sys/kernel/debug/outA/num_samples: 2
/sys/kernel/debug/outA/position: 35435
/sys/kernel/debug/outA/position_sp: 0
/sys/kernel/debug/outA/ramping: N
/sys/kernel/debug/outA/run_direction: 1
/sys/kernel/debug/outA/run_to_pos_active: N
/sys/kernel/debug/outA/run_to_pos_stop_action: 0
/sys/kernel/debug/outA/speed: 0
/sys/kernel/debug/outA/speed_pid_ena: Y
/sys/kernel/debug/outA/stalled: Y
/sys/kernel/debug/outA/state: 0
run
/sys/kernel/debug/outA/dir_chg_samples: 0
/sys/kernel/debug/outA/duty_cycle: 100
/sys/kernel/debug/outA/got_new_sample: N
/sys/kernel/debug/outA/hold_pid_ena: N
/sys/kernel/debug/outA/max_us_per_sample: 100000
/sys/kernel/debug/outA/num_samples: 2
/sys/kernel/debug/outA/position: 35435
/sys/kernel/debug/outA/position_sp: 0
/sys/kernel/debug/outA/ramping: N
/sys/kernel/debug/outA/run_direction: 1
/sys/kernel/debug/outA/run_to_pos_active: N
/sys/kernel/debug/outA/run_to_pos_stop_action: 0
/sys/kernel/debug/outA/speed: 0
/sys/kernel/debug/outA/speed_pid_ena: Y
/sys/kernel/debug/outA/stalled: Y
/sys/kernel/debug/outA/state: 0
```

In testing, sometimes the 3rd trial would run the motor and sometimes it would not. In the example above, it did not. It jumped out at me that `duty_cycle` was `100` since this is the max limit and the motors did not run when `duty_cycle == 100`.

At this point I had already had already read through `set_duty_cycle()` carefully looking for obvious problems, so... 

```
        if (duty_cycle == ev3_tm->duty_cycle)
                return;
```

was fresh in my mind. It occurred to me that if the motor was maxed out at that speed, the duty cycle would never change from 100 and was still at 100 from the last run, therefore we would never tell the motors to turn on later in the function.

---

So, to sum up, kernel debugging tools are `printk()` and debugfs. And you just have to work through things one step at I time until you fully understand what is going on.

rhempel, 2016-11-14:
Yep - prints is a tool of choice. @dlech I am pretty sure that at one point there was some code in the driver that changed the last known speed of the motor for precisely this reason - but the motor drivers have changed once or twice in the last couple of years 😀

dlech, 2016-11-14:
Yes, debugfs should be enabled in the some of the more recent kernel releases. I don't remember exactly when I added it.

dlech, 2016-11-14:
If not, open a new issue and I will make sure it gets turned on.

dlech, 2016-11-16:
Fix released in `17-ev3dev` kernel.

closed 2016-11-16 by dlech