ev3dev/ev3dev #754: EV3 brick permanetly failures to boot [red light]

voxelcam, 2016-10-04:
Hi everyone!

While trying to reformat and make SD-card with last ev3dev I've found that my original EV3 firmware is out-of-date (Installed last Mindstorms Educational at Win7/Virtualbox/Ubuntu Host and choose to Upgrade). Brick began to updating and just turned red led ON forever. 

I tried:
1. All possible reboots: a) regular, b) removing batteries and waiting, c) pressing reset button under 2nd battery, d) 3-buttons reset "back+center+right", e) power-on with 2 buttons, f) power-on with 1 button (both last after reset button).
2. Different a) cables, b) OS'es d) PC
3. Different SD-cards, all with ev3dev (working at other bricks)
4. Finally I'd made LEGO/UART=>USB cable. Pinout used: Tx-Blue(1), Rx-Yel(1), Gnd-Red(4) + second cable only Gnd-Red(4) to SensorPort 4 (because of weak grounding Port 1 sometimes). This cable works with others Brick, shows everything it should. With this brick NOTHING is shown.

To sum up: 
- always red light, except 3-buttons rebooting
- SD-card is not booting at all 
- Brick is not seen by USB at all (like not-connected) under all OS'es and cables
- 3buttons reset just turns off red light and it starts again in 2 seconds
- empty display always
- Port1 UART doesn't not produce any output with working cables

Research:
- EV3 contains EEPROM bootloader which should load u-boot.bin from 0x0 
- Can't find EEPROM dump
- Can't find way to reprogram directly board without using USB/SD-card

What else am I missing? Please help to revive my brick. Can I use some kind of SPI/JTAG and so on to directly reflash it with something of reprogram EEPROM could help? I'm compleately lost at this point.

Kindly regards, Vladimir.

dlech, 2016-10-04:
> pressing reset button under 2nd battery

This is not a reset button! It is used to detect when the rechargeable battery is being used.

> What else am I missing?

What is displayed on the screen when it boots?

If you hold down the right button when turning on the EV3, the screen should say "Updating". When the screen says "Updating", use the official EV3 software to load the firmware.

If the screen doesn't say "updating" or "starting" or you can't get the firmware to load, contact LEGO customer support.

voxelcam, 2016-10-04:
Thank you for your answer!
I didn't NOT disassemble ev3, but had read somewhere that this is reset button (to discharge inside capasitors) - it doesn't matter anyway, because all attempts were made with 1..10 min interval without battaries ( by the way: ONLY fresh Alkaline 6*AA were used for all experiments, no rechargables) and pressing that button without USB and battaries couldn't affect the results.

As I said "empty display always" which means now I can't get ANY on-state pixels on display during all mentioned procedures.

About customer support: I've placed request to LEGO officials last week, but nothing received yet from them.  I wander if they refuse to help me, what can I do by myself - may be I'm looking forward to reprogram EEPROM with new bootloader and internal SPI memory with new u-boot.bin instructions.

voxelcam, 2016-10-04:
I've read [u-boot branch](https://github.com/linaro-swg/u-boot/tree/master/board/lego/ev3) it says:
a) The EV3 contains a bootloader in EEPROM that loads u-boot.bin from address 0x0
of the spi flash memory...
b) The EEPROM contains a program for uploading an image file to the flash memory.
The program is started by holding down the right button on the EV3 when powering
it on....

So it looks like EEPROM fail - it can't do work: after turning on with right button Brick is still without USB so the new firmware can't be uploaded from the USB jack.
Also independent of right button u-boot is not activated because no SD-card reading, no internal memory reading and no USART init at all. 

But: it does something: self-reset (or at least turns red light for 1..2 seconds after long pressing Center+Right+Back)

dlech, 2016-10-04:
One obvious thing you have probably already checked, make sure you are using fully charged batteries.

LEGO has very good customer support and I am sure they will be able to help you. They do receive many requests though, so sometimes they are slow to respond.

voxelcam, 2016-10-05:
dlech, thank you for you answer. I've written in the 2nd message:  "ONLY fresh Alkaline 6*AA were used for all experiments", so it can't be a source of problems. Did you miss it? The same goes with your question dealing with display: "empty display always" - no letters, blinking, separate pixels at all.

EEPROM: everywhere is mentioned that it is ROM and not accessed by regular processes. One [link](http://www.ev3dev.org/docs/kernel-hackers-notebook/ev3-eeprom/) about boot loader place 0x0000-0x3567: Boot loader, it's type: 24FC128 (Microchip [datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/21191s.pdf)). And it says we have to init bluetooth first to avoid I2C readings corruption. According to datasheet there is special WP pin, which disables EEPROM writing when connected to Vcc. 

From [here](http://lists.denx.de/pipermail/u-boot/2016-February/246673.html) : The EV3 itself initally boots from write-protected EEPROM, so no u-boot SPL is needed.

From [here](http://www.ev3dev.org/docs/kernel-hackers-notebook/ev3-eeprom/) we can see how to dump EEPROM from working brick:  
For example reading 1 byte (-n 1) is provided:
hexdump -e '"V0.%02X"' -s 0x3f00 -n 1 < /sys/bus/i2c/drivers/at24/1-0050/eeprom

Hope I will be able to boot ev3dev with another working brick, read EEPROM content to the file, upload it to PC, unsolder 24FC128 from dead EV3, reprogram it, verify and solder back again.

Than I hope the process of reflashing and USB init will work by pressing right button during start-up and I will be able to update whole FLASH regular way.

dlech, 2016-10-05:
> Did you miss it?

Yes, you wrote very much, so read quickly and missed a few things.

---

I really thing you should wait to hear from LEGO before hacking on your EV3. They will very likely send you a new one. If you have modified the hardware, then they may not want to give you a new one.

Also, it could be that the CPU is dead or the RAM is dead or there is a bad power regulator or something else besides a bad EEPROM.

If you do want to try flashing an EEPROM, be aware that there is a bug with the eeprom driver in Linux and this particular EEPROM that causes the first byte read to be corrupt for some reason.

voxelcam, 2016-10-05:
Trying to dump the EEPROM from working brick:
1. [Make a sd-card with ev3dev](http://www.ev3dev.org/docs/getting-started/) (only after special format procedure Etcher worked! )
2. Plug mini-USB and make the [connection to Ubuntu](http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/)
3. [Setup SSH](http://www.ev3dev.org/docs/tutorials/connecting-to-ev3dev-with-ssh/) and go in
4. Trying to read HW revision and MAC from the very end (0x3F00) of the EEPROM

```
robot@ev3dev:~$ hexdump -e '"V0.%02X"' -s 0x3f00 -n 1 < /sys/bus/i2c/drivers/at24/1-0050/eeprom
V0.06
robot@ev3dev:~$ hexdump -e '5/1 "%02X:" "%02X"' -s 0x3f06 -n 6 < /sys/bus/i2c/drivers/at24/1-0050/eeprom
00:16:53:41:51:49

```

5. Dumping the whole EEPROM to screen

```
hexdump -n 16383  < /sys/bus/i2c/drivers/at24/1-0050/eeprom

0000000 4954 4150 5963 5853 590d 5853 0006 0003
0000010 0001 0018 0205 0000 0909 0001 590d 5853
...
0003550 7eb8 8000 0102 0801 0012 0000 7ec4 8000
0003560 5906 5853 4000 8000 ffff ffff ffff ffff
0003570 ffff ffff ffff ffff ffff ffff ffff ffff
*
0003f00 f906 ffff ffff 1600 4153 4951 ffff ffff
0003f10 ffff ffff ffff ffff ffff ffff ffff ffff
*
0003fff

```

everything is nice and neat. Trying to save it to the file:
`cat /sys/bus/i2c/drivers/at24/1-0050/eeprom >eeprom.dump`
and download it to the Ubuntu host using scp:
`scp robot@ev3dev.local:eeprom.dump .`
checking:
```hexdump -e '5/1 "%02X:" "%02X"' -s 0x3f06 -n 6 <eeprom.dump 
00:16:53:41:51:49```
It works! Checking md5sum:
```

md5sum /sys/bus/i2c/drivers/at24/1-0050/eeprom
0b1f342c134b35e93e19d73c7449989d  /sys/bus/i2c/drivers/at24/1-0050/eeprom

```
and local copy (using Ubuntu console):
```
md5sum eeprom.dump 
0b1f342c134b35e93e19d73c7449989d  eeprom.dump
```
So will wait for LEGO to answer. Otherwise will disassemble, unsolder and reprogram memory chip.  My dump from WORKING brick:
[eeprom.zip](https://github.com/ev3dev/ev3dev/files/510158/eeprom.zip)

dlech, 2016-10-11:
I hope you get this fixed. I'm closing the issue since it is not actually a problem with ev3dev.

closed 2016-10-11 by dlech

BertLindeman, 2016-10-11:
Not ev3dev related, and *not* to re-open the discussion . . . 
I like to add my good experience with LEGO support. Entered my complaint just over a week ago and got an email today stating a replacement EV3 brick is in the post.