ev3dev/ev3dev #755: NXT motor recognized as EV3 large motor

WasabiFan, 2016-10-06:
I realized last night while testing things for rhempel/ev3dev-lang-python#227 that the NXT motor I have been using is detected and operated using the `lego-ev3-l-motor` driver instead of the NXT one. Is this expected, and will it cause issues?

dlech, 2016-10-06:
It is expected. I'm not sure if they are different enough to be detected correctly or not. The official EV3 software treats them as the same, but we actually have a slightly different maximum speeds for them in our drivers.

WasabiFan, 2016-10-06:
OK, so this shouldn't cause any issues? Might the detection be inconsistent? If it isn't serving a purpose or the purpose can't be properly served, I imagine it should just be removed.

dlech, 2016-10-06:
BrickPi and PiStorms use NXT motors because that is what they were designed for. On the EV3 though, the current auto-detection only knows about EV3 large and medium motors. It will never detect anything as an NXT motor.

WasabiFan, 2016-10-06:
OK, that makes sense. Can we mention that in the docs somewhere? I think it's important to make it clear that the NXT motor class isn't actually used in the vast majority of cases, so one shouldn't expect to use it in their code on an EV3.

cc @dwalton76 @ddemidov

dlech, 2017-07-28:
This has been documented, so closing.

http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/ev3.html#output-ports

closed 2017-07-28 by dlech