ev3dev/ev3dev #757: I2C sensors that don't work on BrickPi/BrickPi+ and their future

bmegli, 2016-10-12:
We need to add this information somewhere on [ev3dev](http://www.ev3dev.org/) page.

Those devices don't work on BrickPi+/BrickPi regular input ports
- ht-nxt-smux
- mi-xg1300
- lms-absolute-imu
- ms-ev3-smux
- ms-nxtmmx
- ms-8ch-servo
- ms-pps58-nx

They should work on BrickPi [input port 5](http://www.ev3dev.org/docs/ports/brickpi-in-port-5/) but apart from mi-xg1300 were not tested.

 I am not sure if input port 5 has all the pins like regular EV3 inputs so something may not work.

BrickPi+ doesn't have input port 5 at all!

The reason why they don't work is because they are implemented using `sensor_info->ops`. Most of those ops communicate with smbus using Linux I2C adapter driver which we don't have for BrickPi.

It may be possible to implement it for BrickPi and make them work.

2 of the devices (ms_imu, mi_xg1300l) don't use smbus but use ops for post-processing the data. It should be possible to make them work even without I2C adapter driver.

This would need splitting ops to smbus ops and post-processing ops for the 2 devices [here](https://github.com/ev3dev/lego-linux-drivers/blob/master/brickpi/brickpi_i2c_sensor.c#L124).

Alternatively we could just use a flag like `smbus_ops` to be tested [there](https://github.com/ev3dev/lego-linux-drivers/blob/master/brickpi/brickpi_i2c_sensor.c#L124) along `ops` indicating that `ops` use smbus.

We would rather avoid it not to add addditional layer of complexity for just 2 devices or hardcode exceptions for devices but it is an option if we are not going to write I2C adapter driver.

bmegli, 2016-10-12:
So if someone is desperate to use now on BrickPi+:
- ms-absolute-imu
- mi-xg1300l

Let us know here, it looks like a simple change.

bmegli, 2016-10-13:
If you would want to use one of those on BrickPi+:
- ht-nxt-smux
- ms-ev3-smux
- ms-nxtmmx
- ms-8ch-servo
- ms-pps58-nx

Also let us know. Otherwise I don't think anybody will be looking into that anytime soon.

muhe1991, 2017-01-13:
Hey, I am really interested that the sensor "ms-absolute-imu" would work. We need it for our scientific project in the university. Could you please tell me the actual status of it?

bmegli, 2017-01-13:
Just in case you don't know - it works with EV3, should work with BrickPi (only on [input port 5](http://www.ev3dev.org/docs/ports/brickpi-in-port-5/)) and will not work with BrickPi+.

@muhe1991  - do you need it for BrickPi+? If yes and @dlech agrees to accept such solution I could:
- make exception for ms-absolute-imu and mi-xg1300l for BrickPi/BrickPi+
- test it with mi-xg1300l / BrickPi (this is hardware I have, ms-absolute-imu seems to be in the same situation)
- if ev3dev has some generic solution for all those i2c sensors in the future the exception would be reverted

But I would need to know if @dlech would accept temporary solution before I start looking into that.

bmegli, 2017-01-13:
mi-xg1300l uses ops `send_cmd_post_cb`,  `probe_cb`, `remove_cb`
ms-absolute-imu uses ops `send_cmd_post_cb`

Unlike other I2C sensors, mentioned in this issue, their ops implementations don't use i2c_smbus_read/write so they could work with BrickPi/BrickPi+

I believe that it would be sufficent to:

1. Replace [brickpi_i2c_sensor.c line 124](https://github.com/ev3dev/lego-linux-drivers/blob/master/brickpi/brickpi_i2c_sensor.c#L124) with :

``` C
if (sensor_info->ops //temporary workaround for mi-xg1300l and ms-absolute-imu
&& ldev->entry_id->driver_data != MI_CRUIZCORE_XG1300L 
&& ldev->entry_id->driver_data != MS_ABSOLUTE_IMU ) {
```

2. In the same function `brickpi_i2c_sensor_probe` before registering add:

``` C
//temporary workaround for mi-xg1300l and ms-absolute-imu
if (data->info->ops && data->info->ops->probe_cb) {
	err = data->info->ops->probe_cb(data);
	if (err < 0)
		goto err_probe_cb;
}
```

3. In  [brickpi_i2c_sensor_remove](https://github.com/ev3dev/lego-linux-drivers/blob/master/brickpi/brickpi_i2c_sensor.c#L193)  add:
``` C
//temporary workaround for mi-xg1300l and ms-absolute-imu
if (data->info->ops && data->info->ops->remove_cb)
	data->info->ops->remove_cb(data);
```

4. In [brickpi_i2c_sensor_send_command](https://github.com/ev3dev/lego-linux-drivers/blob/master/brickpi/brickpi_i2c_sensor.c#L91) add:

``` C
//temporary workaround for mi-xg1300l and ms-absolute-imu
if (sensor->info->ops && sensor->info->ops->send_cmd_post_cb)
	sensor->info->ops->send_cmd_post_cb(sensor, command);
```

This is mostly what @dlech has implemented for other platforms but couldn't for BrickPi without i2c_smbus_read/write support. This would still work for mi-xg1300l and ms-absolute-imu because they don't use smbus in ops.

Other i2c devices with ops wouldn't get through `probe` so no additional tests seem necessary.

@dlech, would you accept temporary exception for 2 devices?

jabrena, 2017-01-13:
I think that EV3Dev should have a good IMU support.

ms-absolute-imu
mi-xg1300l

If anyone need 2 hands to test that sensors, I could do it.

Cheers

Juan Antonio

dlech, 2017-01-13:
The ideal solution would be to write a real i2c adapter driver for BrickPi. According to <https://github.com/ev3dev/ev3dev/issues/640#issuecomment-223417806>, we can send up to 8 i2c messages at a time with up to 16 bytes of data. These drivers are relatively simple to write. Examples [here](https://github.com/ev3dev/lego-linux-drivers/blob/master/ev3/legoev3_i2c.c) and [here](https://github.com/ev3dev/lego-linux-drivers/blob/master/evb/evb_pru_i2c.c) (the latter one is probably the best example for this case).

To implement it, The master_xfer callback should add messages to a queue and then wait for a completion. Then `brickpi_get_values` needs to be updated so that [the i2c tx part](https://github.com/ev3dev/lego-linux-drivers/blob/master/brickpi/brickpi_ld.c#L222) sends up to 8 messages from this queue. And finally, [the i2c rx part](https://github.com/ev3dev/lego-linux-drivers/blob/master/brickpi/brickpi_ld.c#L299) need to be updated to notify the completion.

But, if @muhe1991 has a specific need on BirckPi, I'm OK with the workaround to get it going sooner.

muhe1991, 2017-01-16:
@bmegli Thanks for the fast reply. We have the BrickPi+ and would love to have a working version of ev3dev for the ms-absolute-imu sensor.

bmegli, 2017-01-16:
@muhe1991 first thing - it's @dlech who really **knows things** here, I am just a humble wanderer who likes to read the code :-) 

I have made a hack that should make ms-absolute-imu work in [pull 58](https://github.com/ev3dev/lego-linux-drivers/pull/58) but it is really ugly and I don't think @dlech will want it in his code unless it's really temporary. Making mi-xg1300l work turned out to be beyond simple hack so it's only for ms-absolute-imu.

You can always follow the dark path - building custom kernel. In that case rpi-krnel with lego linux drivers with hack submodule  from [here](https://github.com/bmegli/lego-linux-drivers/tree/brickpi-i2c).

I am going to work on i2c adapter driver for BrickPi, that @dlech mentioned, in the free time but it will take **time**.

> These drivers are relatively simple to write

Haha, **relatively** is the key word here :wink: how about **relatively** difficult.

bmegli, 2017-01-27:
@muhe1991 

Options for ms-absolute-imu on BrickPi+:
1. Wait (unkown timeframe)
2. Ask @dlech to incorporate workaround temporarilly (unknown answer)
3. Build it yourself (effort)

One way to build it yourself is:

Prerequisities:
- Debian or derivative OS (Ubuntu, Mint, etc.)
- SD card reader or USB SD card reader if your linux is on Virtual Machine

1. Prepare SD card with some recent ev3dev image for RPI (make it work)
2. Follow [ev3dev-buildscripts](https://github.com/ev3dev/ev3dev-buildscripts) first time build instructions replacing step 2 with:

``` bash
cd work
git clone git://github.com/ev3dev/ev3dev-buildscripts
git clone git://github.com/ev3dev/rpi-kernel
```

Edit `.gitmodules` file in `rpi-kernel` replacing module url to `git://github.com/bmegli/lego-linux-drivers`
Then get the workaround code with:
```
cd rpi-kernel
git submodule init
git submodule update
cd drivers/lego
git checkout brickpi-i2c
```

3. Follow the rest of  [ev3dev-buildscripts](https://github.com/ev3dev/ev3dev-buildscripts) first time build instructions

Note that BrickPi doesn't support autodetection so the sensor driver has to be loaded manually, some information in [brickpi-in-port](http://www.ev3dev.org/docs/ports/brickpi-in-port/) and [brickpi-i2c-sensor](http://www.ev3dev.org/docs/drivers/brickpi-i2c-sensor/).

dlech, 2017-02-08:
I have just written a new I2C driver for BrickPi3. It should not be too hard to adapt it to BrickPi.

https://github.com/ev3dev/lego-linux-drivers/blob/master/brickpi3/brickpi3_i2c.c

Basically, this driver just needs to be copied, removed all of the 3's and update the `master_xfer` function to work with BrickPi.

bmegli, 2017-02-08:
This is great news.

I am totally burnt-out and have to solve issues with myself.

dlech, 2017-02-08:
Since ev3dev-jessie is pretty much stable at this point, I am not going to implement the I2C adapter for BrickPi until ev3dev-stretch so that we don't break existing users. I will have another look at @bmegli's workaround for ev3dev-jessie though.

dlech, 2017-02-08:
Workaround for the IMU sensor has been released in 19-ev3dev kernel.

Leaving the issue open with the intention of implementing a proper I2C adapter driver later.

jabrena, 2017-02-21:
hi @dlech, @bmegli,

If I have the sensor: mi-xg1300l connected to Sensor port 1 on a BrickPi+, using the command line, how to read the values from the sensor?

The same solution could be valid for EV3 Brick?

Cheers

Juan Antonio

bmegli, 2017-02-21:
@bmegli wrote:
> Making mi-xg1300l work turned out to be beyond simple hack so it's only for ms-absolute-imu.

**so it's only for ms-absolute-imu**

@dlech wrote:
>Workaround for the IMU sensor has been released in 19-ev3dev kernel.
> Leaving the issue open with the intention of implementing a proper I2C adapter driver later.

**a proper I2C adapter driver later**

@jabrena asked:
> using the command line, how to read the values from the sensor?

From http://www.ev3dev.org/docs/ports/brickpi-in-port/:

>Modes:
nxt-i2c	NXT/I2C sensor

> No sensors are loaded by default. You must manually specify the sensor that is connected and its address by using the set_device attribute. This is equivalent to manually loading I2C devices. The sensor port address will be the BrickPi port address with “:i2c” and the decimal I2C address appended. The [BrickPi I2C sensor page] has information regarding the use of sensors connected to a brickpi-in-port

Then http://www.ev3dev.org/docs/drivers/brickpi-i2c-sensor/ 
And http://www.ev3dev.org/docs/sensors/using-i2c-sensors/
And http://www.ev3dev.org/docs/sensors/mindsensors.com-gyro-multisensitivity-accelerometer-and-compass/

Probably:
1. Set port mode to nxt-i2c (`echo nxt-i2c > somepath/mode`) 
2. Set i2c device to ms-absolute-imu (`echo ms-absolute-imu 0x11 > somepath/set_device`
3. Then as usual (`cat value0`)

bmegli, 2017-02-21:
@jabrena asked:
>The same solution could be valid for EV3 Brick?

You can change the mode manually and set i2c device manually on EV3 if you want for some reason.

On EV3, hardware along with code that @dlech wrote autodetects most sensors (where it is possible)

jabrena, 2017-02-21:
Hi @bmegli, 

many thanks for the instructions, I think that I could do it for ms-absolute-imu.

In relation to Cruizcore sensor, does exist other way to read values?

Cheers

bmegli, 2017-02-21:
> In relation to Cruizcore sensor, does exist other way to read values?

I think that @dlech wrote i2c driver for BrickPi3 so it may already work with BrickPi3, I don't know if rest of his code supports it so you have to wait for his answer.

For ev3dev/BrickPi+/mi-xg1300l apart from waiting for the driver, technically there are other ways ranging from soldering additinal [BrickPi+](https://github.com/DexterInd/BrickPi/tree/master/Hardware%20Design) port with I2C SCL and SDA in parallel to MCP3021; using Dexter Industries libraries directly (without ev3dev or after disabling some kernel modules); modyfying mi-xg1300l driver so that it doesn't support commands, etc.

You are on your own with those however.

jabrena, 2017-02-21:
I will wait for new i2c driver and I will try to finish the support for `ms-absolute-imu` first and later, I will ask again the status. ms-absolute-imu is a good IMU, but I will add the support about mi-xg1300l later.

jabrena, 2017-09-03:
Finally I implemented the support for `ms-absolute-imu` 
It was easy to implement.