ev3dev/ev3dev #758: FYI - the reason why RTL8188CUS (e.g. Edimax EW-7811Un) lags occasionaly

bmegli, 2016-10-13:
## Symptoms

Every once in a while RTL8188CUS lags for around 2 seconds 
The data is not sent during this time, probably stored in network buffers.
Once the lag finishes the data from network buffers is sent in one big batch.

You can experience it for example:
- with ssh - letters not coming either way during 2 sec time and then getting them all at once.
- with ping - from external device where response time rises to around 2 seconds occasionally

It's interesting however that all the packets seem to arrive eventually (unless you exceed network buffer size I guess).

## Hardware

I know only about Edimax EW-7811Un which I own but maybe other devices are affected as well.
If you get this with other hardware let us know.

## How do I know that

I was writing a wireless library recently and saw that I am able to reproduce this behaviour at arbitrary frequency.

## What causes that

Scan trigger sent to wireless device after which a passive or active wireless scan of nearby networks follows.

## How do I reproduce that

You can trigger a wireless scan from one ssh session and observe the symptoms:
- from another session (keep typing while commanding a trigger)
- ping the device from external device (keep pinging while commanding a trigger)

You can trigger a scan with `iw`

Get iw:
``` bash
sudo apt-get update
sudo apt-get install iw
```

Check your wireless interface name:

``` bash
sudo iwconfig
```

To trigger a scan for `wlan0` interface:

```bash
sudo iw dev wlan0 scan trigger
```

In the meanwhile try to type from ssh session or ping from external device.

## Can you do anything about that

Probably yes.

Once your wireless device is associated with Access Point scanning can be disabled.
This will mean however that your device will not switch to another AP even if it has better signal.

In ESS network with multiple BSSes under one SSID (e.g. one WiFi in whole building) this could probably cause your device to not switch between BSSes.

Also scanning can be simply made less often.

This needs further research.

## Finally

Let us know if you are experiencing it also with other hardware (apart from Edimax EW-7811Un )

dlech, 2016-10-13:
Which version of connman are you running? There was some discussion on the connman mailing list sometime in the last year about scanning for other wifi devices after connecting to one. I don't remember what the resolution was, but I think there might have been an option added that we could enable by default.

bmegli, 2016-10-13:
> Which version of connman are you running?

I think that I am up to date:

``` bash
uname -a
Linux ev3dev 4.4.19-15-ev3dev-ev3 #1 PREEMPT Sun Sep 4 16:14:11 CDT 2016 armv5tejl GNU/Linux
```

``` bash
sudo apt-get update
[...]
```

```bash
apt-cache policy connman
connman:
  Installed: 1.33-0ev3dev1
  Candidate: 1.33-0ev3dev1
  Version table:
 *** 1.33-0ev3dev1 0
        990 http://ev3dev.org/debian/ jessie/main armel Packages
        100 /var/lib/dpkg/status
     1.21-1.2 0
        500 http://ftp.debian.org/debian/ jessie/main armel Packages
```

I haven't had time to investigate yet why, only found what ;-)

> There was some discussion on the connman mailing list sometime in the last year about scanning for other wifi devices after connecting to one. I don't remember what the resolution was, but I think there might have been an option added that we could enable by default.

Ok! Thank you. I will try to investigate when I have time.

laurensvalk, 2016-10-18:
@antonvh: This might explain the [RUD] (https://twitter.com/elonmusk/status/743097668725940225) of our balancing robot. Let's try to reproduce this.

antonvh, 2016-10-18:
@laurensvalk Our issue is with bluetooth HID: having it running in the background messes up the control loop timing. Not sure wifi has anything to do with it.

laurensvalk, 2016-10-18:
I'm referring to a different issue that persists even without bluetooth. The robot will operate fine for minutes, when it suddenly falls. As it only happens when WiFi is used, it made me wonder whether this could be it.

bmegli, 2016-10-18:
Note that not all devices work with `iw` (working through nl80211).

For different hardware/driver you may need to use `iwconfig` (through wireless extensions) to trigger a scan.

It's in `wireless-tools` package.

This works for me:

``` bash
sudo apt-get update
sudo apt-get install wireless-tools
sudo iwlist scan
```
Ping drops to 1722 ms while doing that (`iwlist scan` with EW-7811Un device)

bmegli, 2016-10-29:
@antonvh , @laurensvalk you may also want to check if simply gyroscope is not drifting after a few minutes

ev3gyro uses ISZ-655 gyroscope, you can find the specs [here](https://store.invensense.com/datasheets/invensense/PS-ISZ-0655B.pdf)

It has some auto-zero function on pin 21, for compensating the drift, maybe it's called on calibration, you could probably see that in hardware develppers kit schemes of ev3 (or maybe not). 

From ev3dev docs:
> The sensor is calibrated when the GYRO-RATE or the GYRO-G&A mode is set. If the sensor is moving when setting the mode, the calibration will be off

laurensvalk, 2016-10-29:
> @antonvh , @laurensvalk you may also want to check if simply gyroscope is not drifting after a few minutes

That's an unrelated issue. Drift is a slowly varying process, and our code compensates for it. 

What we are seeing is a very sudden interruption of normal program flow which destabilizes the robot. Without WiFi this issue doesn't occur.

laurensvalk, 2016-10-29:
I just realized I have an old video of this issue. [Video] (https://www.youtube.com/watch?v=mEQrW7YfdW0)

In this otherwise unreleased video, the Segway robot is following a line using ev3dev with [this code] (https://github.com/laurensvalk/segway).

At 0:35 into the video, it suddenly falls over.

dlech, 2016-10-29:
is there anything in the kernel log when this happens?

    dmesg | tail

laurensvalk, 2016-10-29:
I'd need to reproduce this to be sure. Right now the robot is somewhere else.

dlech, 2016-11-14:
@bmegli, does this happen with the module parameters set that you mentioned [here](https://github.com/ev3dev/ev3dev/issues/485#issuecomment-252356752)?

bmegli, 2016-11-15:
No, with those:

dlech, 2016-11-15:
FWIW, `rtw_enusbss=0` is already 0 by default, so this setting makes no difference.

In the next kernel release, `CONFIG_POWER_SAVING` will be disabled in the driver, which is the same as `rtw_power_mgnt=0 rtw_ips_mode=0`.

dlech, 2018-07-20:
This is probably caused by ConnMan doing background scanning, which we can turn off:

```
# Enable background scanning. Default is true.
# Background scanning will start every 5 minutes unless
# the scan list is empty. In that case, a simple backoff
# mechanism starting from 10s up to 5 minutes will run.
# BackgroundScanning = true
```

This might also help with connectivity issues with the Visual Studio Code extension.

dlech, 2018-07-20:
More insight on scanning behavior:

https://git.kernel.org/pub/scm/network/connman/connman.git/commit/?id=ac005cefc197a18bb0dcf86497e0a1f5c3bc596b

dlech, 2018-07-20:
Fix released in `ev3dev-connman-config_1.2.0`.

closed 2018-07-20 by dlech