ev3dev/ev3dev #759: FYI - alternative text-to-speech engine that works great with EV3

bmegli, 2016-10-13:
SVOX Pico TTS is alternative TTS engine that has good voice quality compared to espeak.

This is TTS engine that was used in  Android 1.6 “Donut”. It's small.

It supports:
- UK English `en-US`
- US English `en-GB`
- French `fr-FR`
- German `de-DE`
- Spanish  `es-ES`
- Italian  `it-IT`
 
## Try it out

``` bash
sudo apt-get update
sudo apt-get install libttspico-utils
```

### English (default)
``` bash
pico2wave -w=test.wav "Hello there!"
aplay test.wav
```

### German
``` bash
pico2wave -w=test.wav -l de-DE  "Das ist streng verboten!"
aplay test.wav
```

### French
``` bash
pico2wave -w=test.wav -l fr-FR "Bonjour , ma très parfaitement chère ; je suis toute occupée de vous , de votre amitié , de votre santé"
aplay test.wav
```

### Spanish
``` bash
pico2wave -w=test.wav -l es-ES "Hay que estar en silencio!"
aplay test.wav
```

### Italian
``` bash
pico2wave -w=test.wav -l it-IT "Vuole ballare con me?"
aplay test.wav
```

DJuego, 2016-10-29:
Effectively. It is a considerable improvement. It's a pity that do not allow customizing voice. :-/
TTS can seem an accessory value, but it has a lot of impact between  beginners.

Thank you for the suggestion.

DJuego

bmegli, 2016-10-29:
@DJuego 

Yes, it's nice!

Pico TTS is open source but from what I remember those are the only voices that SVOX gives for free.

espeak voice quality is maybe worse but it supports more than 30 languages out of the box.

DJuego, 2016-10-29:
In my case, my target is Spanish. :-P

I have found an interesting idea. Pico TTS can combine with Sox package. It is posible modify the pitch, add effects, etc...  It is really powerful in theory. I only need to invoke the thing from python.

DJuego

closed 2016-11-22 by bmegli

rhempel, 2016-11-22:
Does it allow writing to stdout and then piping to aplay

dlech, 2016-11-22:
Unfortunately, no. But there is a library that goes with it, so you could write your own program quite easily to do this (or modify `pico2wav`).

dlech, 2016-11-22:
Also, the question is still open, should we include this package in the default ev3dev image?

bmegli, 2016-11-22:
@rhempel not out of the box but...

... there are some projects on github that already do that (output to stdout), like here:

https://github.com/DougGore/picopi

bmegli, 2016-11-22:
> Also, the question is still open, should we include this package in the default ev3dev image?

@dlech I don't know, it supports only those 5 languages, but on the other hand those are "major" languages, especially English and Spanish, each with more than 500 million speakers.

Maybe some list of other interesting packages on ev3dev.org without explicit tutorials just a few lines per package?

But I am not sure this is a good idea. This kind of information probably doesn't fit tutorial page, it is not really ev3dev specific, rather linux specific and there is plenty information on the internet already.

pdenoeud, 2019-06-21:
it seems much slower than espeak. Did anybody make any comparison?

pdenoeud, 2019-06-21:
> @rhempel not out of the box but...
> 
> ... there are some projects on github that already do that (output to stdout), like here:
> 
> https://github.com/DougGore/picopi

@bmegli does the fact to forward to stdout make it faster? I find it quite slow when you want to use it to have a voice conversation with a human through a chat bot for instance :-(

bmegli, 2019-06-21:
When compared to writing wav file to sd card, then yes - it might be faster.
I am not sure if ev3dev uses some ramdisk, when the file will be flushed to card and if you will have to wait for it to happen. 

If you are generating wav file and then playing it, you have to wait for the whole speech to be generated before pushing it to output. This is possible place for improvement but only if EV3 is able to generate audio in realtime (fast enough) and pico is able to generate output on the fly.

A lot of questions, as usual, in the end only benchmarks tell the real difference.